carla-simulator/carla #185: How to know all starting points exactly on map?

LoayMAmin, 2018-01-31:
I need to know the starting points near traffic lights (for example) or near cross-roads. Is there a map as to where exactly each point reflects to each position in the map?

hs105, 2018-02-16:
I am interested in this too.

LoayMAmin, 2018-02-16:
I couldn't find a certain way of knowing all the start points visually for each map. So I had to:

1. Make carla start a new episode randomly.

2. Print this number out.

3. Make this number your fixed number from now on instead of starting randomly.

4. In the simlation itself, you can zoom out until you can see where the car is exactly in the map.

5. As per traffic, it turns out that carla only returns ALL the states of all traffic lights in the whole map in each single frame. No specific traffic light that's ahead of you is given. So it's pretty hard to know the excat state of the traffic light ahead of you (the car)

YashBansod, 2018-03-13:
@LoayMAmin why not just open the Carla project in the UE4 Editor and mark the coordinates of all the player start points and see if a traffic light is near it.
Alternatively, map all the player start points to the lane information map and just see its distance from a T intersection. (Traffic lights are only at the intersection)

felipecode, 2018-04-10:
Hello. This was fixed is 0.8.1
Please view http://carla.org/2018/04/05/release-0.8.1/

closed 2018-04-10 by felipecode