carla-simulator/carla #186: Carla crashes on ec2 with protobuf error

dotanrs, 2018-01-31:
When trying to run the compiled version of Carla on aws ec2 ubuntu 16.04, the following error occures

> [libprotobuf FATAL google/protobuf/stubs/common.cc:78] This program was compiled against version 2.6.1 of the Protocol Buffer runtime library, which is not compatible with the installed version (3.3.0).  Contact the program author for an update.  If you compiled the program yourself, make sure that your headers are from the same version of Protocol Buffers as your link-time library.  (Version verification failed in "/build/mir-O8_xaj/mir-0.26.3+16.04.20170605/obj-x86_64-linux-gnu/src/protobuf/mir_protobuf.pb.cc".)
terminating with uncaught exception of type google::protobuf::FatalException: This program was compiled against version 2.6.1 of the Protocol Buffer runtime library, which is not compatible with the installed version (3.3.0).  Contact the program author for an update.  If you compiled the program yourself, make sure that your headers are from the same version of Protocol Buffers as your link-time library.  (Version verification failed in "/build/mir-O8_xaj/mir-0.26.3+16.04.20170605/obj-x86_64-linux-gnu/src/protobuf/mir_protobuf.pb.cc".)

protobuf3 was not installed on the machine when I got this error

nsubiron, 2018-02-01:
I'm not sure why this could happen.

Protobuf 3.3.0 is installed locally by the Setup.sh script, and if everything went well, CARLA is linked against this local version (at "Util/Build/protobuf-install").

dotanrs, 2018-02-01:
I'm using the compiled version from https://github.com/carla-simulator/carla/releases/tag/0.7.1, so maybe it was build with pbf2, but this doesn't fully explain the error since I tried installing the same version on the machine

nsubiron, 2018-02-01:
Protobuf is statically linked, I don't understand why it complains.

dotanrs, 2018-02-02:
Me neither. This is all I did:
- Launch ec2 instance (ubuntu 16.04)
- Copy and unzip compiled Carla
- Fix error of missing openGl*
Then the error appears

* I did it in two different ways on different instances, with the same result.
1: Install X-server per the instructions here http://carla.readthedocs.io/en/latest/how_to_run/
2: install EGL directly as suggested here https://askubuntu.com/questions/244133/how-do-i-get-egl-and-opengles-libraries-for-ubuntu-running-on-virtualbox

nsubiron, 2018-07-30:
Starting from version 0.9.0, we no longer depend on protobuf.

closed 2018-07-30 by nsubiron