carla-simulator/carla #188: Create our own pedestrian models

carla-admin, 2018-02-01:
Create new pedestrian models free to distribute.

dotanrs, 2018-02-13:
Any idea when this will be published? 
If some one wants to add more elements to the simulator, i.e. pets or obstacles (road blocks and etc') on the road can you give some guidelines and explanation how to approach this. 
Please also explain how to generate more weather conditions such as snow and hail.
Thanks!

TheNihilisticRobot, 2018-02-23:
Hi @dotanrs, I'm very sorry I overlooked your question.

Our models are actually almost ready to be added to the project. However, since we are also updating a lot of other things, checking that everything is compatible is taking more time than expected. Hopefully,  we'll have the new release with this and other changes the next week.

TheNihilisticRobot, 2018-02-23:
About creating new content, please, head to this issues: #209 #37 #160 
The Unreal Engine has Its own documentation, there you'll have everything you need to know about creating and placing your own assets. 
Implied in the weather conditions are our [Weather Blueprint](http://carla.readthedocs.io/en/latest/map_customization/) and Unreal's [particle generator](https://docs.unrealengine.com/latest/INT/Resources/Showcases/Effects/).

PR #281 (2018-03-15): Update assets link
#### Description

Updated the assets with the following

  * Fixes #188 Create our own pedestrian models
  * Fixes #235 Remove Automotive Materials dependencies
  * Fixes #255 New cheap background mountains generator
closed 2018-03-15 by nsubiron