carla-simulator/carla #189: Make a CARLA 0.7.1 release for Windows

carla-admin, 2018-02-01:
Make a binary version of CARLA 0.7.1 available to download at [release 0.7.1](https://github.com/carla-simulator/carla/releases/tag/0.7.1).

marcgpuig, 2018-02-05:
Now there is a Carla 0.7.1 binary for Windows ready to [download](https://drive.google.com/file/d/1EMlYKpuZ5UUp680T7qp0gAOU5MyuxBkE/view). Take a look to the [release page](https://github.com/carla-simulator/carla/releases/tag/0.7.1)!

closed 2018-02-05 by marcgpuig

JohnEmad92, 2018-02-06:
I want to ask if i can release a package for windows from ubuntu?

JohnEmad92, 2018-02-06:
Hello @marcgpuig 
I want to ask if I can release a package for windows from ubuntu?

marcgpuig, 2018-02-06:
Unreal Engine does not provide this feature, sorry.
Check out the [Compiling For Linux](https://wiki.unrealengine.com/Compiling_For_Linux) Unreal's wiki page.

JohnEmad92, 2018-02-06:
@marcgpuig 
Thanks for your response
Is the windows release generated from carla on windows?

marcgpuig, 2018-02-06:
@JohnEmad92 
Yes it is!