carla-simulator/carla #190: manual_control.py not working

ajax98, 2018-02-01:
Hi,

I am trying to follow http://carla.readthedocs.io/en/latest/how_to_run/. When I am trying to run  ./manual_control.py --help, I am getting the error that ERROR: (localhost:2000) failed to connect: [Errno 61] Connection refused. What might be the reason for this?

nsubiron, 2018-02-02:
Hi @ajax98,

`./manual_control.py --help` does that? Which version are you running, I'm not able to reproduce.

ajax98, 2018-02-02:
Hi @nsubiron 
I actually run the command ./manual_control.py. I am running the master version.

nsubiron, 2018-02-02:
Hi @ajax98,

Make sure you are running the server with the `-carla-server` argument in the default port, also that you don't have a firewall blocking the connection.

closed 2018-02-12 by nsubiron

KasraMokhtari2112, 2020-04-06:
Hello everyone!

I still have the same issue. Got this error over and over again:
"time-out of 2000ms while waiting for the simulator, make sure the simulator is ready and connected to 127.0.0.1:2000".
I tried the mentioned approaches on 2 different python versions. However, they did not work for me.
I really appreciate it if you can help me with that. Thank you so much!