carla-simulator/carla #194: A simple tutorial for Carla

xfqbuaa, 2018-02-02:
I am Frank from China and very interested in Carla Simulator.
Thanks for Carla team hard work.
I want to do some self-driving car simulation in Chinese traffic scenes based on Carla. 

From my side, a simple tutorial is appreciated for a new user to know:
1, how to set start points
2, how to set goal points
3, how to use map data 
4, a simple straight path planning using existing API 

I spent some time on documents and python client code but still have no idea to integrate my modular pipeline self-driving car algorithm into Carla. 
These tasks may be very basic but really important for a beginner. 
This simple tutorial will benefit for a beginner a lot to know how to use Carla soon.  
Please consider this suggestion.
Thanks a lot.

felipecode, 2018-02-02:
Hey @xfqbuaa.

We are working on improving the documentation.
Thanks for the feedback, we will take this into account

Cheers

AftermathK, 2018-02-03:
@xfqbuaa makes a very good point. When I first built Carla, I edited the original maps and created my own; however, I noticed that during simulation the cars that would spawn would just go around in circles. Are these cars part of Carla's framework or are they part of the UE4? It would be great to figure out how to control their behavior. 
Thank you!

nsubiron, 2018-02-03:
Hi @xfqbuaa, the vehicles are part of CARLA. They follow the road map generated by the _Carla Map Generator_, but you need to trigger the map generation after editing the road by clicking at _"Trigger Road Map Generation"_. The road map created it's serialized together with the road to speed up loading the map. If there is no map, the cars can't find the road and drive in circles. Their behaviour is coded in the `AWheeledVehicleAIController` C++ class.

AftermathK, 2018-02-04:
@nsubiron Thank you for your response. 
Given that the vehicles are part of CARLA and follow the map generated by the Carla Map Generator, if I wanted to completely customize my own roads and wanted to use vehicle models, do you think it would be better to design my own roads using UE4 and then to edit AWheeledVehicleAIController class, or would it be better to edit the Carla Map Generator directly?

nsubiron, 2018-02-05:
@AftermathK If you just want to change the meshes or textures of the road you can reuse our Map Generator. But otherwise this class is not very flexible, it is not going to be easy to add curves, elevation and such.

A workaround for now to use your own roads with the AWheeledVehicleAIController is to set RoutePlanners along your roads (you do so by dragging a spline). When the vehicles enter the trigger box they get a route assigned that they'll try to follow. (RoutePlanner is used for instance for intersections).

For something more complex I'm guessing modifying the AWheeledVehicleAIController is still the easiest.

AftermathK, 2018-02-06:
@nsubiron Very helpful; thank you.
I noticed that on your latest release, it is mentioned that the basic planner is a temporary solution until the map generator is upgraded. What kind of upgrades are you planning on doing to the generator? is this going to be part of the next release or the idea still being developed?

nsubiron, 2018-02-06:
@AftermathK We plan to do major changes to the map generator; curvy roads, elevation, better customization. But this is something to do in the long run as it requires a considerable amount of work.

xfqbuaa, 2018-02-27:
@AftermathK, @felipecode and @nsubiron
Before formal document tutorial, can you share some hints or idea about:
1, python client to set start point and end point
2, how to transfer map data to waypoints during path planning. 

Thanks  a lot.
My goal is to reproduce modular pipeline method in Carla and only 1 scenario with set start point and end point.

albe81x, 2018-03-28:
Hi,
is there any update on how managing trips (start/goal points) and eventually to provide a playback of a registered trip in CARLA? 

Thanks for your support. 
Alberto

felipecode, 2018-04-23:
The driving benchmark module already provides a way to set goal and start points.
Please, check out the new documentation http://carla.readthedocs.io/en/latest/benchmark_start/
We are still improving it. I am sure a way to set start points in command line would be also cool for some quick tests.

albe81x, 2018-04-23:
> en/latest/benchmark_start/

Hi Felipe,
thanks for your reply. I have a question (not directly) related to
benchmarking.
My goal is to define some customized hazard-scenarios (i.e. pedestrian
being hit by the player) to use Carla as a test-bench for ROS algorithms.
In this case I need to keep the other variables (like a pedestrian or
non-player vehicle path) to be always the same to accomplish this task. I
have read in your documentation that Carla does not provide trajectory or
carries any geometric info (like for GPS navigators). Does it mean that
they will always follow a random path or can we preset a pose (start-end
tuple) for a non-player agent in the scenario?

Thanks in advance for your availability and support

Regards,
Alberto

kartik144, 2019-03-14:
> The driving benchmark module already provides a way to set goal and start points.
> Please, check out the new documentation http://carla.readthedocs.io/en/latest/benchmark_start/
> We are still improving it. I am sure a way to set start points in command line would be also cool for some quick tests.

I am not able to access this page. Is the page removed?

nsubiron, 2019-03-14:
@xfqbuaa It was moved to its own repo https://github.com/carla-simulator/driving-benchmarks

eds89, 2019-05-09:
I'm a bit confused on how to use CARLA for training and validation. The best turoail I found was this one: https://medium.com/asap-report/introduction-to-the-carla-simulator-training-a-neural-network-to-control-a-car-part-1-e1c2c9a056a5

Are there plans to release an 'official' tutorial?

liuhe005, 2020-01-11:
@xfqbuaa  how can I set routes for a vehicle ? I am beginner in Carla.

germanros1987, 2020-03-19:
All the problems mentioned in this thread are solved in the current version of CARLA. Please read the documentation for further information: https://carla.readthedocs.io/en/latest

closed 2020-03-19 by germanros1987

KasraMokhtari2112, 2020-05-26:
> All the problems mentioned in this thread are solved in the current version of CARLA. Please read the documentation for further information: https://carla.readthedocs.io/en/latest

I still have the issue with local_planner.py  written for the latest version which is the command can't move the ego car. Have any idea why? the controller is working but It can't move the car at all! I really appreciate your help!