carla-simulator/carla #196: Integrating motion planning algorithms in

MFaisalZaki, 2018-02-02:
Is it possible to integrate motion planning algorithms like RRT* inside CARLA, and draw the tree expansion when running the simulator, I can help with the implementation of RRT* and other motion planning algorithms, but where will be the most appropriate place to implement these algorithms and how it's possible to draw lines in scene from the code, to show the tree expansion.

felipecode, 2018-02-06:
Hello !
Where exactly do you want to show the tree expansion. In a 2D  map or inside the game. For implementing the motion planning in a higher level it is quite straightforward. Basically check the manual_control.py example with the -m option. There you will see all the information you could access, like position on map, lane orientation, other agents position. With this you could program your rrt. We would be happy to see your results ! 

Cheers

MFaisalZaki, 2018-02-06:
I was thinking to integrate tree expansion in the game itself, and my plan is to implement different RRT based algorithms inside CALRA itself to have a variety of motion planning algorithms which can be used by AI cars.

This also can help creating a standard benchmark for RRT based algorithms, where every researcher can use CARLA's integrated motion planning algorithms to check his algorithm performance and then can integrate his work.

What do u think ?

closed 2018-08-06 by stale[bot]