carla-simulator/carla #197: Is engine2learn relevant to easily deploy Carla server inside a docker container ?

jrabary, 2018-02-02:
This project https://github.com/ducandu/MaRLEnE proposes solution to link games developed with unreal to machine learning framework. Especially, it is really interesting to have the possibility to deploy the game engine as a docker container. Is there something similar already coded in Carla or using this project with Carla is relevant ?

nsubiron, 2018-02-05:
Having docker containers for Carla is something we want to have, but it's not there yet. On the other hand, we already have the Python-Unreal communication, what other advantages may give using engine2learn?

jrabary, 2018-02-06:
@nsubiron I'm not very familiar with unreal and how it works under the hood. But I notice that guys from engine2learn use virtual machine and docker container to run the game engine. Also, they don't seem to use VNC at all. That's may be the advantage. This project is linked to another project Tensorforce https://github.com/reinforceio/tensorforce which is a Deep RL library based on TensorFlow and the game engine client behave like a gym environment.

nsubiron, 2018-02-24:
Sorry for the long silence. Yes, how to use Unreal with a docker container is definitively interesting for us, we are too busy as to do it right now but it is in our backlog. I'm going to link this to our issue #203 for making docker containers for CARLA.

Thanks for pointing out this :)

closed 2018-02-24 by nsubiron