carla-simulator/carla #198: carla/carla_server.h not found when building on windows

josheligoldman, 2018-02-02:


nsubiron, 2018-02-03:
Hi @josheligoldman,

Sorry, we don't have yet a reliable way for building CARLA on Windows. You can follow the directions at #21 but for now you'll have to cross check that every step worked.

In this case, the compilation of the CarlaServer module failed; you might need to compile it (either with the Makefile or CMake) and then install it under _"Unreal/CarlaUE4/Plugins/Carla/CarlaServer"_.

josheligoldman, 2018-02-03:
How do I compile the CARLA server. Please provide exact steps

nsubiron, 2018-02-03:
For Linux, the exact steps are in [How to build on Linux](http://carla.readthedocs.io/en/latest/how_to_build_on_linux/). For Windows unfortunately, all we have is issue #21.

marcgpuig, 2018-02-05:
Please consider using the official Carla for Windows release mentioned in issue #189.  
Hope this helps.

closed 2018-02-08 by nsubiron