carla-simulator/carla #199: Problem building on Linux: "pyconfig.h" not found

hmishra2250, 2018-02-03:
When building on Linux, the Setup.sh script is somehow unable to build Boost and I am getting a bunch of errors similar to as show below
```
# include <pyconfig.h>
          ^
1 error generated.

  "clang++" -c -x c++ -fPIC -std=c++1y -stdlib=libc++ -I../llvm-install/include/c++/v1 -O3 -Wno-inline -Wall -pthread -m64 -fPIC -std=c++1y -stdlib=libc++ -I../llvm-install/include/c++/v1 -DBOOST_ALL_NO_LIB=1 -DBOOST_PYTHON_SOURCE -DBOOST_PYTHON_STATIC_LIB -DNDEBUG -I"." -I"/home/him/anaconda3/envs/tensorflow/include/python3.5" -o "bin.v2/libs/python/build/clang-linux-3.9/release/link-static/threading-multi/long.o" "libs/python/src/long.cpp"

...failed clang-linux.compile.c++.without-pth bin.v2/libs/python/build/clang-linux-3.9/release/link-static/threading-multi/long.o...
clang-linux.compile.c++.without-pth bin.v2/libs/python/build/clang-linux-3.9/release/link-static/threading-multi/numeric.o
In file included from libs/python/src/numeric.cpp:6:
In file included from ./boost/python/numeric.hpp:8:
In file included from ./boost/python/detail/prefix.hpp:13:
./boost/python/detail/wrap_python.hpp:50:11: fatal error: 'pyconfig.h' file not found
```

I tried using with and without default python compiler (using Anaconda one and system both), However still am getting the same bunch of errors. Any help or suggestion appreciated.

Thanks.

koryako, 2018-02-05:
the same error

nsubiron, 2018-02-05:
It looks like the Python developer package is not installed, you may try with

    sudo apt-get install python-dev

We don't use boost_python but we are compiling whole boost in the Setup script. I wonder if there is a way of just building the modules we need.

hmishra2250, 2018-02-05:
My bad I didn't update this with my latest findings!

Apparently, as you can see, this is somewhat related to anaconda not being able to find "pyconfig.h", although it's still there. 

Hence the simplest solution is to comment out the anaconda path command from your ~/.bashrc (which causes anaconda's python to override default python). Open a New terminal. Enter the command ```sudo apt-get install python-dev``` to get the latest default python-dev(skip this if u already have one!). Finally, follow the normal install instructions. Boos will be built in place using the provided instructions!

Good Luck! :)

closed 2018-02-20 by nsubiron

ssajj1212, 2018-09-27:
I have the same error. Comment the Anaconda PATH to find pyconfig under your default python path. It works.

AliBaheri, 2018-10-01:
> I have the same error. Comment the Anaconda PATH to find pyconfig under your default python path. It works.

Thanks @ssajj1212 
Could you please give some info how did you comment the Anaconda PATH to find pyconfig under your default python path?

AliBaheri, 2018-10-01:
> My bad I didn't update this with my latest findings!
> 
> Apparently, as you can see, this is somewhat related to anaconda not being able to find "pyconfig.h", although it's still there.
> 
> Hence the simplest solution is to comment out the anaconda path command from your ~/.bashrc (which causes anaconda's python to override default python). Open a New terminal. Enter the command `sudo apt-get install python-dev` to get the latest default python-dev(skip this if u already have one!). Finally, follow the normal install instructions. Boost will be built in place using the provided instructions!
> 
> Don't forget to uncomment Anaconda path in ~/.bashrc to get back your Anaconda working.
> 
> Good Luck! :)

ajboloor, 2018-10-01:
> Could you please give some info how did you comment the Anaconda PATH to find pyconfig under your default python path? Specifically, how can I uncomment Anaconda path in ~/.bashrc?

- Use ``sudo nano ~/.bashrc`` to edit the ~/.bashrc file. 
- Comment the lines starting with `export PATH=...anaconda...`
- Open a new terminal or use ``source ~/.bashrc`` for the changes to have effect.
- Run ``./Setup.sh``

ssajj1212, 2018-10-04:
Hi, @AliBaheri  @ajboloor gives great answer. So after comment the line, just echo $PATH to check whether the python path is correct or not. Once the installation is done, you could uncomment it and use your anaconda.

Derekabc, 2019-06-17:
hi, @hmishra2250  Your answer works well for this problem. One question is that when I comment Anaconda PATH, so I cannot use the Anaconda environment at the same time. But some other programes I have should be run under the Anaconda virtual environment. Any ideas to this problem?

ayadav10491, 2019-12-22:
Deactivating conda base env; commenting "conda initialize" in "bashrc" and Building again worked for me

M-Hammod, 2020-01-12:
you don't need to deactivate conda to build CARLA : 
I solved this issue by finding my pyconfig.h from the command
$python3.7-config --includes --libs

Than export the python lib in ~/.bashrc
will give you the location of python3.7m (if you using ing python3.7 )
gedit ~/.bashrc

export CPLUS_INCLUDE_PATH="$CPLUS_INCLUDE_PATH:~/anaconda3/include/python3.7m"
After this source ~/.bashrc_profile

cheshirekow, 2020-04-16:
Note: `CPLUS_INCLUDE_PATH` works for a plain virtual environment on linux as well. For example, I have an active virtual environment in `/home/josh/.pyenv/python3` so I ran:
```
CPLUS_INCLUDE_PATH="/home/josh/.pyenv/python3/include/python3.6m" make setup
```
and it succeeded

yuan-mingxi, 2024-02-22:
> you don't need to deactivate conda to build CARLA : I solved this issue by finding my pyconfig.h from the command $python3.7-config --includes --libs
>
> Than export the python lib in ~/.bashrc will [...]

i had tried but it still error with"pyconfig.h not found",if there are some other ways to fix?