carla-simulator/carla #200: Define a common interface for easily adding sensors

nsubiron, 2018-02-03:
Make a common interface for cameras and lidar at the server-side. Try to define a general interface for sensors so it's easier to add new sensors in the future.

errorJY, 2018-02-05:
Hello@nusbiron
       What can I do for you？My graduation project is the Sensor expansion for Carla . I will begin to do it after the Spring festival.I think your project will make it  easier，so I want do something for your project.
       please .

errorJY, 2018-02-05:
Hello@nusbiron
What can I do for you？My graduation project is the Sensor expansion for Carla . I will begin to do it after the Spring festival.I think your project will make it easier，so I want to do something for your project.
please .

closed 2018-03-01 by nsubiron

mshaheerm, 2018-06-26:
@errorJY i also need sensor expansion for carla.. have you started working on it?

PR #248 (2018-03-01): Redesign sensor related code

#### Description

  * Fix #200 Define a common interface for easily adding sensors
  * Fix #238 Allow sensor data to be captured asynchronously

> **Note:** Data is not yet captured asynchronously, but the new design allows to easily do it.

#### Important changes

  * _"CarlaSettings.ini"_ syntax changed.
  * Encoding of sensor data through network protocol changed.
  * Small changes to Python API.

#### Where has this been tested?

  - [x] Ubuntu 16.04
  - [x] Windows
  - [x] Python 2.7
  - [x] Python 3.5
  - [x] Unreal Engine 4.17
  - [x] Unreal Engine 4.18

## [... cut]