carla-simulator/carla #203: Docker containers for Carla

nsubiron, 2018-02-05:


volkodava, 2018-03-03:
Hi guys

I did some research on that although end up with the issue described in here: https://github.com/carla-simulator/carla/issues/232

Feel free to use my current solution as a base point for further improvements: https://github.com/volkodava/docker-carla-vnc-desktop

marcgpuig, 2018-03-05:
Hi @volkodava!

We are going to take a look at Docker soon, so your code will be really helpful as a start point.
Thank you for your contribution, we really appreciate it!

Regards!

abmantha, 2018-03-17:
Hi @marcgpuig,

Just wanted to check in on the progress of establishing a stable docker setup. After doing some research, I think Nvidia-Docker (https://github.com/NVIDIA/nvidia-docker) might be the solution to the issue popping up in the rebuild.sh script. As another user reported, Unreal needs to have access to the host GPU. Nvidia has released drivers to solve this issue when spinning up Docker containers.

I will try and take a shot at this over the next week. I'm crunched for time, as my academic research will greatly depend on Carla. 

Would love to know if you're team has made any improvements or garnered any insight into the issue!

Thank you :)

marcgpuig, 2018-03-19:
Hi @abmantha, sorry for the late response.

First of all, thank you for this information related to Nvidia.
Unfortunately we haven't had time to deepen on this topic, although it is an important one for us.

> I will try and take a shot at this over the next week. I'm crunched for time, as my academic research will greatly depend on Carla.

I'm glad to hear that! Get CARLA running on Docker would be awesome. 
If you make some progressions let us know! Our team will be happy to help with all your doubts. 😃

Thanks for your interest in CARLA!

nsubiron, 2019-04-05:
Oh the Docker image has been available since 0.8.3.

closed 2019-04-05 by nsubiron