carla-simulator/carla #204: where is implemention of the function SpawnVehicle() in  VehicleSpawnerBase.h?

barontxu, 2018-02-05:
Hello,
I'm currently reading your awesome code but I find something very confusing that in VehicleSpawnerBase.h, there's a SpawnVehicle() which I can't find the implemention.
Where's it?
Thank you.

nsubiron, 2018-02-05:
Hi @barontxu, good catch ;) the function is implemented in the derived _"VehicleSpawner"_ Blueprint; note that it is marked as `UFUNCTION(BlueprintImplementableEvent)`. When you call this function you are triggering a blueprint event.

The C++ code decides when and where the vehicle is spawned, but the Blueprint decides the model and color of the vehicle. This way the C++ code doesn't need to know about the assets.

barontxu, 2018-02-05:
@nsubiron get it. I saw a SpawnVehicle node in the blueprint, it's it from GetWorld()?

barontxu, 2018-02-05:
@nsubiron Yes i see. Do we have to spawn actor in blueprint? I tried c++, but the car i spawned just can't move.

barontxu, 2018-02-05:
@nsubiron And thanks much to your swiftly respond. You're so nice.

nsubiron, 2018-02-05:
It should work in C++ too. Just make sure the vehicle is a subclass of `ACarlaWheeledVehicle` and follow all the steps in `AVehicleSpawnerBase::SpawnVehicleAtSpawnPoint()` function (you need to spawn the controller and give it a road map).

barontxu, 2018-02-05:
@nsubiron you mean, for example, I can do it with 
GetWorld()->SpawnActor<ACarlaWheeledVehicle >(.....)
?

nsubiron, 2018-02-05:
Yes, that should work

barontxu, 2018-02-05:
Thank you. I'll give it a try.