carla-simulator/carla #208: Carla windows Release

AminYounes92, 2018-02-06:
I'm trying to use the latest release of CARLA on windows and run it as client and server but there i had this error log.
ERROR: (10.200.1.157:2000) failed to connect: [WinError 10061] No connection could be made because the target machine actively refused it
 i wonder if this windows release work normally as client/server connection or not.

marcgpuig, 2018-02-06:
Hi @AminYounes92,

It should work normally, but seems that the game reset (on a new episode) takes too long on Windows and the client throws this error. A quick fix could be to increase the ```timeout``` on the client side (by default is 15 sec).

Try something like:
```py
with make_carla_client(host, port, timeout=1800) as client:
```

AminYounes92, 2018-02-06:
Hi @marcgpuig 

Thank you for your response.
I've tried your proposed solution but still have the same error. I think its more related to how can we run the server side on windows as in LINUX there is a certain command in the terminal to run carla as server which is missing in windows.

marcgpuig, 2018-02-06:
@AminYounes92 
In Windows you can run the server with the same commands as you do in Linux.
Remember to use ```-carla-server``` to enable the server mode.

This works for me:

    CarlaUE4.exe -carla-server -benchmark -fps=5 -windowed -ResX=800 -ResY=600

Remember, the start may be slow, give it some time and then run the python client.

We are currently working on this _experimental_ Windows release, all the feedback and things you can figure it out will be very appreciated :)

marcgpuig, 2018-02-06:
@AminYounes92 
Remember that you can change the timeout in the server side too from the config file, this may help.

Try increasing that value:
```ini
; Time-out in milliseconds for the networking operations.
ServerTimeOut=10000
```

eds89, 2018-02-06:
Hi. Wonderful! But, loading settings via ini file isn't working.

CarlaUE4.exe  -carla-settings="./CARLA_0.7.1\CarlaSettings.ini"

And the ini contents:
; Example of settings file for CARLA.
;
; Use it with `./CarlaUE4.sh -carla-settings=Path/To/This/File`.

[CARLA/Server]
; If set to false, a mock controller will be used instead of waiting for a real
; client to connect.
UseNetworking=true
; Ports to use for the server-client communication. This can be overridden by
; the command-line switch `-world-port=N`, write and read ports will be set to
; N+1 and N+2 respectively.
WorldPort=2000
; Time-out in milliseconds for the networking operations.
;ServerTimeOut=10000
ServerTimeOut=10000
; In synchronous mode, CARLA waits every frame until the control from the client
; is received.
SynchronousMode=true
; Send info about every non-player agent in the scene every frame, the
; information is attached to the measurements message. This includes other
; vehicles, pedestrians and traffic signs. Disabled by default to improve
; performance.
SendNonPlayerAgentsInfo=false

[CARLA/LevelSettings]
; Path of the vehicle class to be used for the player. Leave empty for default.
; Paths follow the pattern "/Game/Blueprints/Vehicles/Mustang/Mustang.Mustang_C"
PlayerVehicle=
; Number of non-player vehicles to be spawned into the level.
NumberOfVehicles=30
; Number of non-player pedestrians to be spawned into the level.
NumberOfPedestrians=60
; Index of the weather/lighting presets to use. If negative, the default presets
; of the map will be used.
WeatherId=1
; Seeds for the pseudo-random number generators.
SeedVehicles=123456789
SeedPedestrians=123456789

[CARLA/SceneCapture]
; Names of the cameras to be attached to the player, comma-separated, each of
; them should be defined in its own subsection. E.g., Uncomment next line to add
; a camera called MyCamera to the vehicle

 Cameras=MyCamera

; Now, every camera we added needs to be defined it in its own subsection.
[CARLA/SceneCapture/MyCamera]
; Post-processing effect to be applied. Valid values:
;   * None                  No effects applied.
;   * SceneFinal            Post-processing present at scene (bloom, fog, etc).
;   * Depth                 Depth map ground-truth only.
;   * SemanticSegmentation  Semantic segmentation ground-truth only.
PostProcessing=SemanticSegmentation
; Size of the captured image in pixels.
ImageSizeX=800
ImageSizeY=600
; Camera (horizontal) field of view in degrees.
CameraFOV=90
; Position of the camera relative to the car in centimeters.
CameraPositionX=15
CameraPositionY=0
CameraPositionZ=123
; Rotation of the camera relative to the car in degrees.
CameraRotationPitch=8
CameraRotationRoll=0
CameraRotationYaw=0

; Stereo setup example:
;
 [CARLA/SceneCapture]
 Cameras=MyCamera,CameraStereoLeft/RGB,CameraStereoRight/RGB
 ImageSizeX=800
 ImageSizeY=600
 CameraFOV=90
[CARLA/SceneCapture/CameraStereoLeft]
 CameraPositionX=170
 CameraPositionY=-30
 CameraPositionZ=150
 CameraRotationPitch=0
 CameraRotationRoll=0
 CameraRotationYaw=0
 [CARLA/SceneCapture/CameraStereoLeft/RGB]
 PostProcessing=SceneFinal
; [CARLA/SceneCapture/CameraStereoLeft/Depth]
; PostProcessing=Depth
 [CARLA/SceneCapture/CameraStereoRight]
 CameraPositionX=170
 CameraPositionY=30
 CameraPositionZ=150
 CameraRotationPitch=0
 CameraRotationRoll=0
 CameraRotationYaw=0
 [CARLA/SceneCapture/CameraStereoRight/RGB]
 PostProcessing=SceneFinal
; [CARLA/SceneCapture/CameraStereoRight/Depth]
; PostProcessing=Depth

marcgpuig, 2018-02-07:
@eds89,
Using the provided file hierarchy:

[image]

Try to modify and use the ```Example.CarlaSettings.ini``` with the settings you gave to me and try the following command:

    CarlaUE4.exe -carla-settings=Example.CarlaSettings.ini
This works fine for me.

nsubiron, 2018-02-08:
Just noticed that absolute paths do not work in Linux either, it's not even showing the _"Unable to find settings file"_ error. I will open an issue for this.

closed 2018-03-22 by nsubiron