carla-simulator/carla #211: Carla's Geometric/Graph Model

AftermathK, 2018-02-07:
Hi all, 

I am currently working on customizing my own maps, but at the same time I want to make sure that Carla can interact with the maps that I design. 
After I create a model for my map (geometric model), will I have to manually generate a graph that the A* algorithm can run on? I read the paper, but I have not being able to determine(and locate the code that can help me with this) if it will be possible to create a complex map (with multiple roads branching off an intersection) that can interact with Carla. Is this possible?

Thanks.

felipecode, 2018-02-08:
Hello @AftermathK ,

Regarding your first question, yes. Unfortunately you will have to manually
generate the graph. For now, CARLA lacks a more complete way to generate a city
and to have all the information sent to the planner. This is something we intend
to solve for Milestone "More flexible generation of Maps"
Also, regarding your second question. No it is not possible to create more complex maps
but this is a high priority issue for us.

closed 2018-02-08 by felipecode

AftermathK, 2018-02-08:
Thank you for your reply @felipecode 
I understand that at the moment Carla's map generator is limited on how customizable it may be, but assuming that I am able to design my own maps in UE4, where may I find the source code that interacts with the graph model that I need to create manually? Do you have any suggestions on creating the graph model manually? because I would probably need to find a way to interact with the planner after the graph is created.

I think these are the main challenges that I am coming across. Any help or guidance is truly appreciated.

Thanks.

Raj270892, 2020-02-26:
hi @AftermathK  how do you get map information in the form of graph(edges and nodes)? I want to see if I can implement A* on map in CARLA