carla-simulator/carla #213: Bad Steering on Carla

Leonardo092, 2018-02-07:


TheNihilisticRobot, 2018-02-07:
Hi @Leonardo092 

Check your steering configuration on the front wheel blueprint of your vehicle, maybe a different steer angle gets you the desired results. Between 60 and 80 generally works for us. Make sure the rear wheels blueprint have the steer angle set to 0, otherwise they will steer with the front wheels.

Your wheels sinking to the ground means the shape radius setted up in the Wheel blueprint doesn't match the vehicle's wheel model. You can adjust it until the wheels seem right.

Check our [Documentation](http://carla.readthedocs.io/en/latest/how_to_add_assets/) to see how we configured our vehicles.

Leonardo092, 2018-02-07:
Hi there. Thanks for replying. Rear angle is properly set to 0 and Front one is 70. No matter how much I vary it (believe me, I tried) the car still keeps on drifting.
Concerning the sinking issue, my blueprint radius matches the physical one. I read about increasing it, but it does not work whatsoever :(

Leonardo092, 2018-02-07:
Maybe a detailed workflow might help:
- In BLENDER I joined several meshes into what is "Vehicle_Base" in the documentation and then set a hierarchy in between it and the four wheels.
- I exported the fbx model created from blender into UE4 (I assume I did it properly, as axes in my PhAT fits the requirements of "X forward, Z up")
- I followed the procedure explained in the Documentation

Worth pointing out: in PhAT, wheels are modeled as spheres and the body as a box centred in its COM as established at BLENDER level

TheNihilisticRobot, 2018-02-07:
I'm not really sure how to identify this issue.
Since you created yourself the Vehicle hierarchy I would suggest using the rig of one of our vehicles and weighting your model to it instead.

Leonardo092, 2018-02-07:
I see... is it possible the to "append" my vehicle's meshes onto one of your models?

TheNihilisticRobot, 2018-02-07:
Sure is. I don't know how to do it in Blender, but if you see one of our vehicle Fbx You can see it is made from a model attached to a skeleton (Rig). You can detach this model, attach (Weight) your own and export it again (With the name of your vehicle).

Leonardo092, 2018-02-08:
I exported one vehicle's skeletal mesh (Mustang's) from UE4 as FBX and then opened it in Blender. You can refer to the figure below:
[image]
My question is: what am I supposed to keep and what to discard from this hierarchy? I tried to take the whole Mustang object over with mine and make it fit the existent skeleton, but when importing it back on UE4 I am facing balance and motion problems.

TheNihilisticRobot, 2018-02-12:
All right, you are on the right track. 
Just keep everything under vertex group and [asign (Weight) each one of them](https://docs.blender.org/manual/en/dev/modeling/meshes/properties/vertex_groups/assigning_vertex_group.html) to the part of the vehicle their names refer to. Export after you've done that and then try importing it again into Unreal.

Leonardo092, 2018-02-15:
This seems to work badly for me. I am running in unknown issues there is no tutorial for, so I decide to change my approach. I want to use EXACTLY one of the available models on UE4 and mounting on it the meshes of mine. Since I want the physics not to be affected by my vehicle, I would like to render my materials while providing them with no mass and no collisions. Is it possible?

PS If I export a model from UE4 to Blender and then I import it back again, its weight is messed up (I made no modification and import/export is made via FBX binary 7.4)

TheNihilisticRobot, 2018-02-16:
Okay sorry I can't help you with Blender but I have never worked with it.
As a last resort, please, send me a link to your vehicle's OBJ or FBX I'll see what the problem might be and try to build a Interchangeable Blender/3DMax/Maya pipeline.

closed 2018-02-20 by Leonardo092