carla-simulator/carla #214: Camera PostProcessing SemanticSegmentation generates void images

hs105, 2018-02-07:
Using PostProcessing = 'SemanticSegmentation', it generates void (black) images. 

I noticed on the documentation page, the latest code generates black images too: here:

http://carla.readthedocs.io/en/latest/cameras_and_sensors/

nsubiron, 2018-02-07:
Hi @hs105,

The semantic segmentation image has the tags encoded in the red channel, these tags go from 0 to 12, that's very dark shades of red. The image in the documentation is a sample, if you take a closer look it is not black but it has some dark shades of red.

closed 2018-02-14 by nsubiron

hs105, 2018-02-16:
Thanks for your clarification.

RogerAylagas, 2018-10-04:
Knowing that, how do you transform the "black image" to the color one?

marcgpuig, 2018-10-05:
Hi @RogerAylagas 
This is an already answered question, but you can do something like this:
```py
from carla import image_converter

# other stuff...

encoded_sem_seg = sensor_data.get('CameraSemSeg', None)
colored_sem_seg = image_converter.labels_to_cityscapes_palette(encoded_sem_seg)
```

kk2491, 2019-01-30:
@marcgpuig I am using the latest pre-compiled version of Carla. I am not able to import ```image_converter```.

```
>>> from carla import image_converter
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: cannot import name image_converter
>>> 
```

Whether the pre-compiled version doesnt support image_converter ?

Thank you,
KK

marcgpuig, 2019-02-11:
@kk2491 which version are you using?

kk2491, 2019-02-12:
@marcgpuig I am using the latest version 0.9.2 pre-compiled one.