carla-simulator/carla #216: Improvements in the benchmarking

felipecode, 2018-02-08:
Currently the system to benchmark AI methods using CORL2017 tests is very long.
It takes around 15 hours more or less. For now it is possible to stop and resume your benchmark, but
there are a few issues on the implementation. 
When the methods restart it tends to overwrite the last task done by the car.
This issues have already been fixed and are under review for version 0.8

mvpcom, 2018-02-21:
Haha, I just wanted to remind you for fixing this and I found this one :D 
For myself, I just removed the question in the Benchmark code for now and the new code is as follows: 

```
def _continue_experiment(self, continue_experiment):
        if self._experiment_exist():
            if continue_experiment:
                line_on_file = self._get_last_position()
	    else:
		line_on_file = 0
        else:
            line_on_file = 0

        return line_on_file
```

mvpcom, 2018-02-21:
Unfortunately, the question itself is a problem because, in the middle of the test, a query will be popped up and need someone to answer it otherwise a connection error will be rise.

PR #341 (2018-04-20): Some benchmark fixes

#### Description

This pull request is a complete redefinition of the benchmark module.
It now has a different and documented architecture and it was more 
extensively tested.
It also corrected some problems when restarting the benchmark
and some problems related to the computed infractions.

Fixes #336 Fixes #216 Fixes #243 Fixes #306
Avoid issues #230, #310

#### Where has this been tested?

python 2.7, 3.5
Ubuntu 16.06
Carla 0.8.1

#### Possible Drawbacks
The old version of the benchmark is not compatible anymore

---
This change is [[image]](https://reviewable.io/revi [... cut]
closed 2018-04-20 by nsubiron