carla-simulator/carla #218: When to release the methods evaluated in the CARLA paper?

tomchen1000, 2018-02-13:
I wondered if you have an update on when to release the methods evaluated in the CARLA paper. Namely the high level topological planner and the three approaches: modular pipeline, imitation learning and reinforcement learning.

felipecode, 2018-02-14:
Hellon @tomchen1000 
The high level topological planner is released on version 0.7.1 

The imitation learning approach is released under the carla-simulator organization .
https://github.com/carla-simulator/imitation-learning

Cheers.

hs105, 2018-02-16:
Thanks. what's your schedule for the other two methods?

felipecode, 2018-02-27:
We hope to release then soon !
Thanks for the interest .

zdx3578, 2018-03-14:
@tomchen1000 @hs105  ref
https://github.com/NervanaSystems/coach
https://github.com/18605973470/rl-with-carla
https://github.com/kvasnyj/carla

eds89, 2018-05-24:
Hi.
By running the evaluated methods, how can I see the car navigating and deviating from obstacles,  while the experiment is executed?

closed 2018-08-06 by stale[bot]

varunjammula, 2019-07-02:
Any plans on the modular pipeline?