carla-simulator/carla #219: Decodifying depth map gives strange result

iamatrashbin, 2018-02-14:
Hello, thank you so much for the good work with Carla. I have a question about the documentation about decoding RGB-depth images to a depth map. In the documentation is says:

>  The order from less to more significant bytes is R -> G -> B. [...]
>  1. To decodify our depth first we get the int24.
`R + G*256 + B*256*256`

However, if I follow that instruction and write:
`depth_rgb = cv2.imread(depth_filename)`
`depth_not_norm = np.add(depth_rgb[:,:,0], np.add(np.multiply(depth_rgb[:,:,1], 256), np.multiply(depth_rgb[:,:,2], 256**2)))`
`depth = np.multiply(np.divide(depth_not_norm, 256**3-1), 1000)`

 I get the quite curious result:
[image]

However, if I do:
`R*256*256 + G*256 + B`
I get a result which looks much better to me:
[image]

Is this wrong in the documentation or have I misunderstood something?

marcgpuig, 2018-02-14:
Hi @iamatrashbin,

I think you are using BGRA, not RGBA.
That could explain everything.

Please take a look at the documentation page [Cameras and Sensors](http://carla.readthedocs.io/en/latest/cameras_and_sensors/#cameras-and-sensors).

> The images are sent by the server as a BGRA array of bytes. The provided Python client retrieves the images in this format, it's up to the users to parse the images and convert them to the desired format. There are some examples in the PythonClient folder showing how to parse the images.

Tell me if it works :)

gs14iitbbs, 2018-08-23:
Hi @iamatrashbin and @marcgpuig,

As I understand that RGB values(24 bits) have to be used to get the depth values for each pixel, as given on [http://carla.readthedocs.io/en/latest/cameras_and_sensors/#camera-depth-map]

But when I load the .npy depth map created by Carla, it has only single channel.

I use the following python script:
```
import numpy as np
from matplotlib import pyplot as plt

img_array = np.load("Town01/1513192538.853153_8_70/depths/1513192540.177777.npy")

print(img_array.shape)
print(img_array.min(), img_array.max())
plt.imshow(img_array)
plt.show()
```
Which give me following output:
`
(600, 800) // single channel ?? why

(5.9604648328104514e-05, 1000.0)
`
[image]

Can anyone tell me how can I extract the RGB components to finally get the depth values in meters?

Thanks in advance!!

marcgpuig, 2018-08-23:
Hey @gs14iitbbs.
Are you 100% sure you are correctly storing the depth images?
I can't help you if I don't know how are you getting the depth image.

Check out this line in [manual_control.py](https://github.com/carla-simulator/carla/blob/stable/PythonClient/manual_control.py) and follow the image process!
https://github.com/carla-simulator/carla/blob/2d52245d3e67dd58454d963887cd4fc3f5ef02fb/PythonClient/manual_control.py#L198

botcs, 2020-07-14:
Unfortunately neither RGB nor BGR works for me. Both results in something that can be seen in the first figure of #219 

I have used the code from `image_converter.py` and still getting the same result.
Any idea what may I have missed?

TzabarDolev, 2023-09-19:
There is an small error in the formula which is given in the 'readthedocs' of the depth sensor (wrong order of color channels).
I wrote about it the CarlaSimBlog - https://carlasimblog.wordpress.com/2023/09/16/sensors/