carla-simulator/carla #220: Adding C++ classes to CARLA

razza-tu, 2018-02-15:
**Question**

What is the intended mechanism for creating my own classes in CARLA? 

**Context**

Currently, I am opening the CARLA project with the UE4 Editor, navigating to the desired folder, and using UE4's "Create C++ Class" option. When doing so, a (suspiciously) long process starts, and I am eventually told: 

> Successfully added class 'LIDAR', however you must recompile the 'Carla' module before it will appear in the Content Browser. Failed to generate project files.

The output log reads as follows:

`LogLinux: Setting swap interval to 'Immediate'
Running /home/rory/UnrealEngine_4.17/Engine/Binaries/DotNET/UnrealBuildTool.exe  -makefile -kdevelopfile -qmakefile -cmakefile -codelitefile  -project="/home/rory/carla/Unreal/CarlaUE4/CarlaUE4.uproject" -game -engine -progress
Running Mono...
~/UnrealEngine_4.17/Engine ~/carla/Unreal/CarlaUE4
Discovering modules, targets and source code for project...
LogSlate: Took 0.000074 seconds to synchronously load lazily loaded font '../../../Engine/Content/Slate/Fonts/Roboto-Regular.ttf' (155K)
Discovering modules, targets and source code for project...
Discovering modules, targets and source code for project...
Discovering modules, targets and source code for project...
Discovering modules, targets and source code for project...
UnrealBuildTool Exception: System.ArgumentOutOfRangeException: startIndex cannot be larger than length of string.
Parameter name: startIndex
  at System.String.Substring (Int32 startIndex, Int32 length) <0x7f0f08c5b780 + 0x000f4> in <filename unknown>:0 
  at System.String.Substring (Int32 startIndex) <0x7f0f08c5b750 + 0x0001e> in <filename unknown>:0 
  at UnrealBuildTool.CodeLiteProject.WriteProjectFile (System.Collections.Generic.List`1 InPlatforms, System.Collections.Generic.List`1 InConfigurations) <0x41b264d0 + 0x00615> in <filename unknown>:0 
  at UnrealBuildTool.ProjectFileGenerator.WriteProjectFiles () <0x41b26350 + 0x000b0> in <filename unknown>:0 
  at UnrealBuildTool.ProjectFileGenerator.GenerateProjectFiles (System.String[] Arguments) <0x41a2e5f0 + 0x01ff4> in <filename unknown>:0 
  at UnrealBuildTool.UnrealBuildTool.GuardedMain (System.String[] Arguments) <0x419de000 + 0x02ffd> in <filename unknown>:0 
LogLinux: Setting swap interval to 'Immediate'
LogLinux: Setting swap interval to 'Immediate'`

**Additional Information**
OS: Ubuntu 16.04 LTS
CPU: Intel® Core™ i7-4770 CPU @ 3.40GHz × 8 
GPU: GeForce GTX 760 (192-bit)/PCIe/SSE2
RAM: 16GB
Unreal Engine: 4.17
CARLA: 0.7.1 (I think)

nsubiron, 2018-02-16:
Hi @razza-tu,

This seems to be an issue of Unreal Editor, I experienced similar issues in Linux. If you recompile and launch the editor again it should be able to find any classes in your cpp files (in fact you don't need to create the class from within the editor, you can just drop it in the source directory).

You can force a recompile by deleting intermediate folders, launching the "Rebuild.sh" script should work too.

razza-tu, 2018-02-19:
Hi @nsubiron,

Thanks for your help so far. I have now:

- Created `[myclass].cpp` and `[myclass].h` in `~/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla`

- Run Rebuild.sh

This has not resulted in [myclass] appearing in the editor. Am I missing something obvious?

Thanks again in advance!

nsubiron, 2018-02-24:
Hi @razza-tu,

Is it possible that you don't have the _"Show plugin content"_ active in the content browser? 
See https://github.com/carla-simulator/carla/issues/180#issuecomment-361199548.

razza-tu, 2018-02-26:
Hi @nsubiron,

Indeed, I did not have the _Show plugin content_ option active in the content browser. Since reading your reply, I have

- Enabled that option

- Closed CARLA

- Run Rebuild.sh

The classes still do not appear. Is there anything else you can think of?

nsubiron, 2018-03-05:
Hi @razza-tu,

We are not able to reproduce this.

Which version of Unreal are you using? We found an issue in 4.18, `GenerateProjectFiles.sh` crashed not generating the Makefiles, thus you might be using an older version of the Makefiles without the new class. Perhaps it also affected 4.17 and we didn't noticed. It should be fixed in master branch now.

To be sure you can remove all intermediate folders inside the Carla plugin, then opening the project directly in Unreal with

```bash
$ cd Unreal/CarlaUE4/Plugins/Carla
$ rm -Rf Binaries Build Intermediate DerivedDataCache
$ cd ../..
$ ~/UnrealEngine_4.18/Engine/Binaries/Linux/UE4Editor "$PWD/CarlaUE4.uproject"
```

razza-tu, 2018-03-05:
Hi @nsubiron ,

Firstly, I'd like to confirm that I am using Unreal 4.17, as recommended.

Secondly, I have now followed your instructions:

- The contents of `~/carla/Unreal/CarlaUE4/Plugins/Carla` now reads as follows:

 - `CarlaServer/`

 - `Content/`

 - `Resources/`

 - `Source/`

 - `Carla.uplugin`

- The last command brings up an Unreal Engine dialogue box, telling me that that Carla is incompatible with my Unreal version. I was asked if I wanted to disable the Carla plugin, and upon declining, I was asked if I wanted to rebuild `libUE4Editor-Carla.so`

 - I modified the command to read: `~/UnrealEngine_4.17/Engine/Binaries/Linux/UE4Editor "$PWD/CarlaUE4.uproject"`

 - I ran this command from `~/carla/Unreal/CarlaUE4`

 - Attempting the Unreal prompted rebuild resulted in failure

nsubiron, 2018-03-05:
Just to clarify, `master` branch has been updated to UE4.18 while `stable` branch is still at UE4.17.

Can you share what it prints in the output?

simmates, 2018-03-06:
Has anyone used Qtcreator or Eclipse to write code? In Qtcreator I always get strange warnings and errors from the analyzer.

nsubiron, 2018-03-07:
I haven't tried those, but Unreal generates a lot of code for reflection that is very difficult for any analyzer to parse it properly.

Since 4.18 you have Visual Studio Code on every platform though https://twitter.com/unrealengine/status/930812641068572672

closed 2018-03-14 by nsubiron