carla-simulator/carla #223: Camera frames not showing up

eds89, 2018-02-16:
Hello,

When I load CARLA as a standalone game (non-server), the associated camera frames are not showing up. I can only see them when I launch CARLA as server and connect to it via either Python scripts. It is unclear to me if I should expect seeing the frames for each of the cameras defined in the .ini file. 

I'm loading CARLA via the following command line.
CarlaUE4.exe  -carla-settings=CarlaSettings.ini -windowed -ResX=800 -ResY=600

And, my current settings file can be found below. Comments were removed.

[CARLA/Server]
UseNetworking=false
WorldPort=2000
ServerTimeOut=10000
SynchronousMode=true
SendNonPlayerAgentsInfo=false
[CARLA/LevelSettings]
PlayerVehicle=
NumberOfVehicles=90
NumberOfPedestrians=150
WeatherId=1
SeedVehicles=123456789
SeedPedestrians=123456789
[CARLA/SceneCapture]
Cameras=MyCamera
[CARLA/SceneCapture/MyCamera]
PostProcessing=Depth
ImageSizeX=512
ImageSizeY=384
CameraFOV=90
CameraPositionX=15
CameraPositionY=0
CameraPositionZ=123
CameraRotationPitch=8
CameraRotationRoll=0
CameraRotationYaw=0

eds89, 2018-02-16:
Hello,
Apparently, the settings file isn't being loaded up, according to the log file:

`LogInit: Command line:  -windowed -carla-settings=`

marcgpuig, 2018-02-16:
Hi @eds89,

> It is unclear to me if I should expect seeing the frames for each of the cameras defined in the .ini file.

No, you shouldn't.
The server does only display the main camera only for debug purposes.

The reason that you see the other cameras in the client side is because it is coded in the [`manual_control.py`](https://github.com/carla-simulator/carla/blob/master/PythonClient/manual_control.py) to display these frames in that certain position in the screen.

eds89, 2018-02-16:
So, isn't there a way to see what each camera is recording while the car
runs through the level?

marcgpuig, 2018-02-19:
@eds89,

> isn't there a way to see what each camera is recording

If you mean in the **server**, it is not implemented because this is not a necessary feature for us.
Anyway, as it is a post processing material, you can attach it to a certain camera to visualize it.

It's up to you use the Unreal Engine to render different cameras in different frames on a single buffer.

Regards.

eds89, 2018-02-19:
Hello,
That's ok. I've managed to change the python code to load settings
dynamically (on manual_driving.py). I'm now looking for instructions on how
to save recorded images as png files.

marcgpuig, 2018-02-23:
Hi @eds89,

Sorry for the late response.
Please, make sure you have took a look at the provided python examples before asking.

In the [`client_example.py`](https://github.com/carla-simulator/carla/blob/master/PythonClient/client_example.py) you can find an example of saving the images:

https://github.com/carla-simulator/carla/blob/69f7e8ea68046340b1174e8c867381721852b8e1/PythonClient/client_example.py#L104-L107

Good luck :)

closed 2018-03-14 by marcgpuig