carla-simulator/carla #227: Where Are the Cameras? How to Add More?

josheligoldman, 2018-02-17:
Hi,
Where are the cameras placed on the car? How do I access them using the python client? How do I add more cameras?

marcgpuig, 2018-02-19:
Hi @josheligoldman,

You can add them from the python client when creating the `CarlaSettings` object that will be sent to the server.
Check out these lines of [`client_example.py`](https://github.com/carla-simulator/carla/blob/master/PythonClient/client_example.py):

https://github.com/carla-simulator/carla/blob/69f7e8ea68046340b1174e8c867381721852b8e1/PythonClient/client_example.py#L55-L71

You have another example in [`manual_control.py`](https://github.com/carla-simulator/carla/blob/master/PythonClient/manual_control.py#L71-L96)

Regards :)

closed 2018-03-14 by marcgpuig

MuhammadSafwan456, 2020-11-12:
Hi @josheligoldman ,
you can add camera RGB sensor in Carla by using 
```
camera_bp = blueprint_library.find("sensor.camera.rgb")
camera_bp.set_attribute("image_size_x", f"{img_widht}")    
camera_bp.set_attribute("image_size_y", f"{img_height}")
camera_bp.set_attribute("fov",f"{90}")
```

here we define the camera blueprint and other properties like image width , height and field of View.
now if you want to spawn the camera on any vehicle so then you need to spawn a vehicle first.
Vehicle is spawned by following code
```
vehicle_bp = blueprint_library.filter('vehicle.audi.a2')[0]
spawn_point = random.choice(world.get_map().get_spawn_points())
vehicle = world.spawn_actor(vehicle_bp , spawn_point)
```

Now we will define the spawn point for camera which will be relative to vehicle position

`spawn_point_car = carla.Transform(carla.Location(   x = 0.6 , y = -0.45 , z = 1.6 ) )
`
the x y and z parameters in Carla.Location refer to the forward/Backward , Left/Right and Up/Down position of camera relative to the vehicle

`camera_right = world.spawn_actor(camera_bp , spawn_point_right, attach_to = vehicle)`

now camera is spawned and attached to our vehicle.

now to access the image output of camera we can do this

camera.listen(lambda image : image.save_to_disk("/Location/Where-You/Want-to/Store-Image/Image-Name.png")