carla-simulator/carla #229: Release of 'Straight' Environment

bhairavmehta95, 2018-02-19:
Would there be any way for the team to release the `straight` environment from the benchmark, almost as a sanity check?

felipecode, 2018-02-19:
hey @bhairavmehta95 ,

The straight task is already released. It is the first task from the benchmark (Check [benchmark](http://carla.readthedocs.io/en/latest/benchmark/) ). Currently
we have just two environments, CARLA TOWN 01 and CARLA town 02. Straight,
is just a task on these environment where the agent has a goal right in front of it.

Regards.

bhairavmehta95, 2018-02-19:
Thank you for getting back so quickly.

So if I wanted to train an RL agent on this straight task, I would redefine the `Manual` class in https://github.com/carla-simulator/carla/blob/master/PythonClient/run_benchmark.py?

closed 2018-02-19 by bhairavmehta95