carla-simulator/carla #230: benchmark repetition problem

TairuiWang, 2018-02-19:
Hi, I am currently using the Carla benchmark to test my algorithm. My Carla version is 0.7.1. 
I modified on 'CoRL2017.py'. I changed the start-end points in function 'def _poses_straight():'. Then, I found that if, for example, I set the point pairs as [[36, 40], [36, 40], [36, 40], [36, 40]], the first episode and the second are always different. From the second episode to the rest, they are the same. I printed out the world location of these points pairs and they are the same.
The command I used to launch the Carla is:

> ./CarlaUE4.sh /Game/Maps/Town01 -carla-server -benchmark -fps=15 -windowed -ResX=800 -ResY=600

Any idea how can I fix it? Thanks a lot.

felipecode, 2018-02-20:
Hey @TairuiWang 

Thanks for the issue.
I will try to reproduce it soon ! 
Seems that it may be a bug.

Cheers.

felipecode, 2018-04-20:
Please check issue #310
The episodes positions change after the first reset.
We wont fix this for now ( There is a turnaround when running the benchmark), 
but this problem wont exist in the next CARLA major version, since we are going to change the API.

closed 2018-04-20 by felipecode