carla-simulator/carla #231: Client Object

bhairavmehta95, 2018-02-20:
Is there any way to pass a `client` object and assign it to something so I don't have to have everything in one big file?

I understand that the yields would make this impossible, but is there any other way to set up the code besides:

```python
with make_carla_client(host, port) as client:
    # all of your other code
```

bhairavmehta95, 2018-02-20:
Never mind, the `manual_control.py` would work.

bhairavmehta95, 2018-02-22:
For any one who was interested, I did find a way around this (thanks to the CARLA team for building some great software).

Here's an excerpt of my code:

```python

while True:
            try:
                self.client = CarlaClient(host, port)
                self.client.connect()
                break
            except TCPConnectionError as error:
                print(error)
                time.sleep(1)

```

For anyone who's more interested on the RL potential of this simulator, I've been doing my best to add OpenAI Gym-esque environments [here](https://github.com/bhairavmehta95/carla-environments)

closed 2018-02-22 by bhairavmehta95

nsubiron, 2018-02-23:
Hi @bhairavmehta95,

I don't understand your issue, in both cases you get a `client` variable that you can pass around. I would prefer the first code because it disconnects the server once it's done, even if an exception is raised.

Thanks for sharing your repo btw :)