carla-simulator/carla #234: Carla in UE4 opening with a lot of Load Errors. (Can't find file for asset)

YashBansod, 2018-02-20:
I faced the issue #131 while running the **./Setup.sh** script in Carla. And I followed the suggestion mentioned by @nsubiron. After manually downloading the **Content_Latest.tar.gz** from the gdrive and extracting its contents in **{carla}/Unreal/CarlaUE4/Content/** , my **./Setup.sh** script and the **./Rebuild.sh** script ran successfully.
But the UE4Editor opened with a lot of Load Errors like this:
[image]

Also when I try to link the automotive materials manually, the following step in documentation:

> Under the "Details" pane, search for the "Parent" material, and replace "DummyCar" material by "M_Carpaint" material.

I cannot find the M_Carpaint material in search.

nsubiron, 2018-02-20:
Hi @YashBansod,

You need to download yourself the automotive materials and extract them in the Content folder. Please follow the instructions in [How to add the automotive materials](http://carla.readthedocs.io/en/latest/how_to_add_automotive_materials/).

YashBansod, 2018-02-20:
Also after the 

> Launching UE4Editor...
part in the ./Rebuild.sh I get a lot of messages like these:

> [2018.02.20-08.55.20:072][  0]LogLinker: Warning: Can't find file '/Game/AutomotiveMaterials/Materials/Glass/MI_Glass'
> 1001[2018.02.20-08.55.20:072][  0]LogLinker: Warning: Can't find file for asset '/Game/AutomotiveMaterials/Materials/Glass/MI_Glass' while loading ../../../../carla/Unreal/CarlaUE4/Content/Static/Vehicles/GeneralMaterials/MaterialRedirectors/MI_Glass_Redirector.uasset.
> [2018.02.20-08.55.20:073][  0]LoadErrors: Error: /Game/Static/Vehicles/GeneralMaterials/MaterialRedirectors/MI_Glass_Redirector : Can't find file for asset. /Game/AutomotiveMaterials/Materials/Glass/MI_Glass
> 1001[2018.02.20-08.55.20:073][  0]LogLinker: Warning: Can't find file '/Game/AutomotiveMaterials/Materials/Glass/MI_Glass'
> 1001[2018.02.20-08.55.20:074][  0]LogLinker: Warning: Can't find file for asset '/Game/AutomotiveMaterials/Materials/Glass/MI_Glass' while loading ../../../../carla/Unreal/CarlaUE4/Content/Static/Vehicles/GeneralMaterials/MaterialRedirectors/MI_Glass_Redirector.uasset.
> [2018.02.20-08.55.20:074][  0]LoadErrors: Error: /Game/Static/Vehicles/GeneralMaterials/MaterialRedirectors/MI_Glass_Redirector : Can't find file for asset. /Game/AutomotiveMaterials/Materials/Glass/MI_Glass
> 1001[2018.02.20-08.55.20:627][  0]LogLinker: Warning: Can't find file '/Game/Static/Buildings/ResidentialTallWStairs/N2/_ResidentialTallWStairs_GroundFloor_Door_Mat_N2_OcclusionRoughnessMetallic'
> 1001[2018.02.20-08.55.20:630][  0]LogLinker: Warning: Can't find file for asset '/Game/Static/Buildings/ResidentialTallWStairs/N2/_ResidentialTallWStairs_GroundFloor_Door_Mat_N2_OcclusionRoughnessMetallic' while loading ../../../../carla/Unreal/CarlaUE4/Content/Static/Buildings/ResidentialTallWStairs/N2/ResidentialTallWStairs_Door_N2_Inst.uasset.
> [2018.02.20-08.55.20:630][  0]LoadErrors: Error: /Game/Static/Buildings/ResidentialTallWStairs/N2/ResidentialTallWStairs_Door_N2_Inst : Can't find file for asset. /Game/Static/Buildings/ResidentialTallWStairs/N2/_ResidentialTallWStairs_GroundFloor_Door_Mat_N2_OcclusionRoughnessMetallic
> 1001[2018.02.20-08.55.20:644][  0]LogLinker: Warning: Can't find file '/Game/Static/Buildings/ResidentialTallWStairs/N2/_ResidentialTallWStairs_GroundFloor_Door_Mat_N2_OcclusionRoughnessMetallic'
> 1001[2018.02.20-08.55.20:646][  0]LogLinker: Warning: Can't find file for asset '/Game/Static/Buildings/ResidentialTallWStairs/N2/_ResidentialTallWStairs_GroundFloor_Door_Mat_N2_OcclusionRoughnessMetallic' while loading ../../../../carla/Unreal/CarlaUE4/Content/Static/Buildings/ResidentialTallWStairs/N2/ResidentialTallWStairs_Door_N2_Inst.uasset.
> [2018.02.20-08.55.20:646][  0]LoadErrors: Error: /Game/Static/Buildings/ResidentialTallWStairs/N2/ResidentialTallWStairs_Door_N2_Inst : Can't find file for asset. /Game/Static/Buildings/ResidentialTallWStairs/N2/_ResidentialTallWStairs_GroundFloor_Door_Mat_N2_OcclusionRoughnessMetallic
> 1001[2018.02.20-08.55.40:468][  0]LogLinker: Warning: Can't find file '/Game/AutomotiveMaterials/Materials/Glass/MI_Glass'
> 1001[2018.02.20-08.55.40:470][  0]LogLinker: Warning: Can't find file for asset '/Game/AutomotiveMaterials/Materials/Glass/MI_Glass' while loading ../../../../carla/Unreal/CarlaUE4/Content/Static/Props/TelephoneBox/Prop_TelephoneBox.uasset.
> [2018.02.20-08.55.40:470][  0]LoadErrors: Error: /Game/Static/Props/TelephoneBox/Prop_TelephoneBox : Can't find file for asset. /Game/AutomotiveMaterials/Materials/Glass/MI_Glass
> 1001[2018.02.20-08.55.40:484][  0]LogLinker: Warning: Can't find file '/Game/AutomotiveMaterials/Materials/Glass/MI_Glass'
> 1001[2018.02.20-08.55.40:485][  0]LogLinker: Warning: Can't find file for asset '/Game/AutomotiveMaterials/Materials/Glass/MI_Glass' while loading ../../../../carla/Unreal/CarlaUE4/Content/Static/Props/TelephoneBox/Prop_TelephoneBox.uasset.
> [2018.02.20-08.55.40:485][  0]LoadErrors: Error: /Game/Static/Props/TelephoneBox/Prop_TelephoneBox : Can't find file for asset. /Game/AutomotiveMaterials/Materials/Glass/MI_Glass
> 1001[2018.02.20-08.55.42:218][  0]LogLinker: Warning: Can't find file '/Game/AutomotiveMaterials/Materials/Glass/MI_Glass'
> 1001[2018.02.20-08.55.42:219][  0]LogLinker: Warning: Can't find file for asset '/Game/AutomotiveMaterials/Materials/Glass/MI_Glass' while loading ../../../../carla/Unreal/CarlaUE4/Content/Static/Vehicles/4Wheeled/AudiA2_/Vh_Car_AudiA2_Static_Norig.uasset.
> [2018.02.20-08.55.42:219][  0]LoadErrors: Error: /Game/Static/Vehicles/4Wheeled/AudiA2_/Vh_Car_AudiA2_Static_Norig : Can't find file for asset. /Game/AutomotiveMaterials/Materials/Glass/MI_Glass
> 1001[2018.02.20-08.55.42:229][  0]LogLinker: Warning: Can't find file '/Game/AutomotiveMaterials/Materials/Glass/MI_Glass'
> 1001[2018.02.20-08.55.42:230][  0]LogLinker: Warning: Can't find file for asset '/Game/AutomotiveMaterials/Materials/Glass/MI_Glass' while loading ../../../../carla/Unreal/CarlaUE4/Content/Static/Vehicles/4Wheeled/AudiA2_/Vh_Car_AudiA2_Static_Norig.uasset.
> [2018.02.20-08.55.42:230][  0]LoadErrors: Error: /Game/Static/Vehicles/4Wheeled/AudiA2_/Vh_Car_AudiA2_Static_Norig : Can't find file for asset. /Game/AutomotiveMaterials/Materials/Glass/MI_Glass
> 1001[2018.02.20-08.55.42:275][  0]LogLinker: Warning: Can't find file '/Game/AutomotiveMaterials/Materials/Glass/MI_Glass'
[... cut ...]
> 1001[2018.02.20-08.55.42:778][  0]LogLinker: Warning: Can't find file '/Game/AutomotiveMaterials/Materials/Glass/MI_Glass'
> 1001[2018.02.20-08.55.42:779][  0]LogLinker: Warning: Can't find file for asset '/Game/AutomotiveMaterials/Materials/Glass/MI_Glass' while loading ../../../../carla/Unreal/CarlaUE4/Content/Static/Vehicles/4Wheeled/VolkswagenT2/Vh_Van_VolkswagenT2_NOrig.uasset.
> [2018.02.20-08.55.42:779][  0]LoadErrors: Error: /Game/Static/Vehicles/4Wheeled/VolkswagenT2/Vh_Van_VolkswagenT2_NOrig : Can't find file for asset. /Game/AutomotiveMaterials/Materials/Glass/MI_Glass
> 
>

YashBansod, 2018-02-20:
@nsubiron my bad. I added the automotive materials to **{carla}/Unreal/CarlaUE4/** instead of **{carla}/Unreal/CarlaUE4/Content**. Will try rebuilding it.

YashBansod, 2018-02-20:
I placed the Automotive materials folder in the Content folder and again ran the ./Rebuild.sh script. Still the following errors were thrown:

> [2018.02.20-10.52.28:742][  0]LogLinker: Warning: Can't find file '/Game/Static/Buildings/ResidentialTallWStairs/N2/_ResidentialTallWStairs_GroundFloor_Door_Mat_N2_OcclusionRoughnessMetallic'
> 1001[2018.02.20-10.52.28:744][  0]LogLinker: Warning: Can't find file for asset '/Game/Static/Buildings/ResidentialTallWStairs/N2/_ResidentialTallWStairs_GroundFloor_Door_Mat_N2_OcclusionRoughnessMetallic' while loading ../../../../carla/Unreal/CarlaUE4/Content/Static/Buildings/ResidentialTallWStairs/N2/ResidentialTallWStairs_Door_N2_Inst.uasset.
> [2018.02.20-10.52.28:744][  0]LoadErrors: Error: /Game/Static/Buildings/ResidentialTallWStairs/N2/ResidentialTallWStairs_Door_N2_Inst : Can't find file for asset. /Game/Static/Buildings/ResidentialTallWStairs/N2/_ResidentialTallWStairs_GroundFloor_Door_Mat_N2_OcclusionRoughnessMetallic
> 1001[2018.02.20-10.52.28:749][  0]LogLinker: Warning: Can't find file '/Game/Static/Buildings/ResidentialTallWStairs/N2/_ResidentialTallWStairs_GroundFloor_Door_Mat_N2_OcclusionRoughnessMetallic'
> 1001[2018.02.20-10.52.28:751][  0]LogLinker: Warning: Can't find file for asset '/Game/Static/Buildings/ResidentialTallWStairs/N2/_ResidentialTallWStairs_GroundFloor_Door_Mat_N2_OcclusionRoughnessMetallic' while loading ../../../../carla/Unreal/CarlaUE4/Content/Static/Buildings/ResidentialTallWStairs/N2/ResidentialTallWStairs_Door_N2_Inst.uasset.
> [2018.02.20-10.52.28:751][  0]LoadErrors: Error: /Game/Static/Buildings/ResidentialTallWStairs/N2/ResidentialTallWStairs_Door_N2_Inst : Can't find file for asset. /Game/Static/Buildings/ResidentialTallWStairs/N2/_ResidentialTallWStairs_GroundFloor_Door_Mat_N2_OcclusionRoughnessMetallic
> 1001[2018.02.20-10.52.37:430][  0]LogLinker: Warning: Can't find file '/Game/Static/Buildings/ResidentialTallWStairs/N2/_ResidentialTallWStairs_GroundFloor_Door_Mat_N2_OcclusionRoughnessMetallic'
> [2018.02.20-10.52.37:430][  0]LoadErrors: Failed to load /Game/Static/Buildings/ResidentialTallWStairs/N2/_ResidentialTallWStairs_GroundFloor_Door_Mat_N2_OcclusionRoughnessMetallic._ResidentialTallWStairs_GroundFloor_Door_Mat_N2_OcclusionRoughnessMetallic Referenced by ResidentialTallWStairs_Door_N2_Inst
> 1001[2018.02.20-10.52.37:432][  0]LogLinker: Warning: Can't find file for asset '/Game/Static/Buildings/ResidentialTallWStairs/N2/_ResidentialTallWStairs_GroundFloor_Door_Mat_N2_OcclusionRoughnessMetallic' while loading NULL.

and

> [2018.02.20-10.53.52:370][  0]LogMaterial: Warning: Material /Game/AutomotiveMaterials/Materials/Glass/M_Glass.M_Glass needed to have new flag set bUsedWithInstancedStaticMeshes !
> [2018.02.20-10.53.52:388][  0]LogMaterial: Missing cached shader map for material M_Glass, compiling. 
> [2018.02.20-10.53.52:488][  0]LogMaterial: Missing cached shader map for material M_Glass, compiling. 
> [2018.02.20-10.53.58:019][  0]LogScript: Warning: Accessed None trying to read property Floor
> 	MutlipleFloorBuilding_C /Game/Maps/Town01.Town01:PersistentLevel.MutlipleFloorBuilding7
> 	Function /Game/Static/Buildings/MutlipleFloorBuilding.MutlipleFloorBuilding_C:UserConstructionScript:0210
> [2018.02.20-10.53.58:021][  0]LogScript: Warning: Accessed None trying to read property Floor
> 	MutlipleFloorBuilding_C /Game/Maps/Town01.Town01:PersistentLevel.MutlipleFloorBuilding6
> 	Function /Game/Static/Buildings/MutlipleFloorBuilding.MutlipleFloorBuilding_C:UserConstructionScript:0210
> [2018.02.20-10.53.58:025][  0]LogScript: Warning: Accessed None trying to read property Floor
> 	MutlipleFloorBuilding_C /Game/Maps/Town01.Town01:PersistentLevel.MutlipleFloorBuilding4_5700
> 	Function /Game/Static/Buildings/MutlipleFloorBuilding.MutlipleFloorBuilding_C:UserConstructionScript:0210

When the UE4Editor opens, it shows this window: 
[image]

Is this supposed to happen?

YashBansod, 2018-02-20:
Also, the even though I followed all the steps in [Manually link the materials](http://carla.readthedocs.io/en/latest/how_to_add_automotive_materials/), I am getting the car without the paint:
[image]

nsubiron, 2018-02-20:
About the missing materials after linking, it is a known issue (#18).

And the other errors that appear are also known, @TheNihilisticRobot is working on that. Hopefully we will remove these in the coming release, but for now it is safe to ignore them.

YashBansod, 2018-02-20:
Thanks @nsubiron, your provided solution in #18 worked.

closed 2018-02-20 by YashBansod