carla-simulator/carla #243: Benchmark Result for Table 2 of CORL2017 Paper

mvpcom, 2018-02-23:
I found that in the current benchmark code [corl_2017.py](https://github.com/carla-simulator/carla/blob/master/PythonClient/carla/benchmarks/corl_2017.py) information about the second table (Table 2: Average distance (in kilometers) traveled between two infractions) is not generated as the result. 

I found the following function in your code, and it seems to me it is under developing, yeah? 
```
def _get_details(self):
        # Function to get automatic information from the experiment for writing purposes
        return 'corl2017_' + self._city_name
```

Here is the result of my experiment after running benchmark by your saved model and with this result I don't think if I can reproduce the second table using this information only:
```
Metric :  average_completion
  Weather:  4.0
    Task  0  ->  0.982042053485
    Task  1  ->  0.95359464593
    Task  2  ->  0.980107061204
    Task  3  ->  0.949471559897
    AvG  ->  0.966303830129
  Weather:  14.0
    Task  0  ->  0.983389089245
    Task  1  ->  0.936733661307
    Task  2  ->  0.882419393675
    Task  3  ->  0.756894498771
    AvG  ->  0.88985916075
Metric :  intersection_otherlane
  Weather:  4.0
    Task  0  ->  0.0
    Task  1  ->  0.0
    Task  2  ->  0.0
    Task  3  ->  0.0
    AvG  ->  0.0
  Weather:  14.0
    Task  0  ->  0.0
    Task  1  ->  0.0
    Task  2  ->  0.0
    Task  3  ->  0.108286684898
    AvG  ->  0.0270716712246
Metric :  average_fully_completed
  Weather:  4.0
    Task  0  ->  1.0
    Task  1  ->  0.92
    Task  2  ->  0.96
    Task  3  ->  0.875
    AvG  ->  0.93875
  Weather:  14.0
    Task  0  ->  1.0
    Task  1  ->  0.875
    Task  2  ->  0.727272727273
    Task  3  ->  0.625
    AvG  ->  0.806818181818
Metric :  intersection_offroad
  Weather:  4.0
    Task  0  ->  0.0
    Task  1  ->  0.0
    Task  2  ->  0.0
    Task  3  ->  0.0
    AvG  ->  0.0
  Weather:  14.0
    Task  0  ->  0.0
    Task  1  ->  0.0
    Task  2  ->  0.0
    Task  3  ->  0.0541433424492
    AvG  ->  0.0135358356123
Metric :  collision_other
  Weather:  4.0
    Task  0  ->  0.0
    Task  1  ->  0.0
    Task  2  ->  0.0
    Task  3  ->  0.0619189096224
    AvG  ->  0.0154797274056
[... 260 lines cut ...]
    Task  0  ->  0.0
    Task  1  ->  0.0
    Task  2  ->  0.0
    Task  3  ->  0.205866190553
    AvG  ->  0.0514665476384
  Weather:  3.0
    Task  0  ->  0.0
    Task  1  ->  0.0
    Task  2  ->  0.0
    Task  3  ->  0.272665197274
    AvG  ->  0.0681662993185
  Weather:  6.0
    Task  0  ->  0.0
    Task  1  ->  0.0
    Task  2  ->  0.0
    Task  3  ->  0.383565953556
    AvG  ->  0.0958914883891
  Weather:  8.0
    Task  0  ->  0.0
    Task  1  ->  0.0
    Task  2  ->  0.0
    Task  3  ->  0.241075648212
    AvG  ->  0.0602689120529

```

TairuiWang, 2018-02-26:
Hi @mvpcom , can I know that how long did you spend on running their benchmark? It seems their benchmark is longer than 24 hours. BTW, my gpu is 1080Ti.

mvpcom, 2018-02-26:
@TairuiWang For the Town02 it was less than 15Hours, but for Town01 I can confirm that it was more than 22 hours.

felipecode, 2018-02-26:
@mvpcom  Thanks ! For the issue.  You are right. 
It is about 24 hours for Town 01 and 15 for Town 02.

Regarding the issue.
Yes,
We are still working on the benchmark. 
I am going to work in these issues in a few weeks.

Cheers.

TairuiWang, 2018-03-05:
Hi guys, 

Thanks very much for your reply!
Another question is that what the fps setting did you use? I set the frame rate as 10Hz. But my tested result is not as good as their paper presented. BTW, the paper I mean is [this](https://arxiv.org/abs/1711.03938) one. 

For example. I tested the trained model on 'New Town' with 'One turn'. I got 48% success rate. But the result from paper is 59%. 

Best,
Tairui

TairuiWang, 2018-03-08:
@felipecode Thanks for your great work! Could you give me more details about how to get 'infraction analysis' result? I want to test my own model with that.
Here are some questions:
1) how to get the driving distance? In my opinion, one way is: speed * time_interval. Another way is calculate euclidean distance between the two locations.
2) What's your criteria of collision and off-road? Did you set any threshold of percentage to decide the failure?
3) Did you use the same testing path for your 'infraction analysis' as the 'Dynamic navigation' in benchmark?

Finally, if it is possible, could you share a short script so that I can modify it and reproduce the result of 'infraction analysis' of your model. This can help me to find out the missing implementation details from your paper.

Thank you very much!
Tairui wang

felipecode, 2018-03-20:
Hello @TairuiWang 

Sorry for late answer.
Here it goes.
1. Euclidean distance does not work. The way to go is to accumulate distance traveled on every
step. Otherwise it wouldn't count when the agent go back to the same point, for instance.

2. This is done in some arbitrary way. There are some parameters specified in the code. 
We need to set like a threshold in the impact and also collision time. The results could
change depending on this threshold.
Offroad on the other hand is simple. We consider a car going offroad if it is stays more than
30% of it is area out of the road. After the car is considered offroad, it counts another offroad
for every 30 seconds it stays outside of the road. 

3. Yes

Thanks for the question.
I will be analyzing the infraction analysis in the next few days to see what is wrong. 

Felipe Codevilla.

closed 2018-04-20 by nsubiron

felipecode, 2018-04-20:
@TairuiWang , @mvpcom 
This new pull request #341  of the  should fix  the driving benchmark problems. Please take a look !

PR #341 (2018-04-20): Some benchmark fixes

#### Description

This pull request is a complete redefinition of the benchmark module.
It now has a different and documented architecture and it was more 
extensively tested.
It also corrected some problems when restarting the benchmark
and some problems related to the computed infractions.

Fixes #336 Fixes #216 Fixes #243 Fixes #306
Avoid issues #230, #310

#### Where has this been tested?

python 2.7, 3.5
Ubuntu 16.06
Carla 0.8.1

#### Possible Drawbacks
The old version of the benchmark is not compatible anymore

---
This change is [[image]](https://reviewable.io/revi [... cut]