carla-simulator/carla #245: Unable to get the NPC Cars get spawned.

YashBansod, 2018-02-24:
I want to create a scenario where there is only one NPC car in the entire MAP and the Player CAR spawns just behind it. Later I plant to write a code for making the Player CAR follow the NPC.

For doing this I duplicated the Town01 Map of Carla in UE4 Editor and Removed all the PlayerStarts except 1. Then I duplicated the PlayerStart and placed it in front of the current one. **This is all the change I did in the Town01 Map**. Now I start the simulator by hitting the play button, but only the player car gets spawned and no NPC cars get spawned.
Later I put 6 more PlayerStarts but still NPC Cars won't spawn. Am I doing something wrong? Do I understand correctly that the NPC Cars also spawn in the PlayerStart locations?

[image]

nsubiron, 2018-02-24:
Hi @YashBansod,

Yes, the NPC vehicles also spawn on PlayerStarts, but if there is not enough PlayerStarts to spawn the requested amount of NPC vehicles, none is spawned. I see this is not indeed the most intuitive.

You should change you CarlaSettings `NumberOfVehicles=2`. Also make sure that there is enough space between PlayerStarts to spawn the vehicle.

Sorry, the vehicle spawner is a bit buggy right now, I'm going to add an issue on this to our backlog.

YashBansod, 2018-02-24:
@nsubiron thanks for the reply. I am unable to find where I should set the NumberOfVehicles. I am not running a Server Client Session, I am just running the simulator from the UE4 Editor.
Also, Is there a reason that the player always spawns at the latest created PlayerStart?

YashBansod, 2018-02-24:
Thanks a lot @nsubiron. This worked. I know that you guys are working on developing the documentation for Carla and in fact, if I were to use just the precompiled version of Carla, the documentation is more than enough to help you get started. But really the documentation related to modifying or even understanding Carla in the UE4 Editor is scarce. It would be a great help if you could create a small documentation explaining the basic structure of Carla in UE4 Editor and some basic overview. This can just be a video that you guys can make of a screen capture while explaining the structure of Carla and how to make minor modifications.

closed 2018-02-24 by YashBansod