carla-simulator/carla #247: How to build the game binaries with only one map?

YashBansod, 2018-02-24:
I made a bare-bones map and named it Town03, with only a road, landscape and a few player start points. 
Now I want to build the binaries so that I can run the simulator in a PC where UE4 Editor isn't installed. But when I try to cook the CarlaUE4 project for Linux, It starts cooking it with all the Towns and the entire simulator and takes a lot of time. 
My build is a very bare-bones and I want to cook the game so that only the Town03 is included while cooking the game.
This will keep the binary size small and really speed up the cooking process.

nsubiron, 2018-02-24:
Hi @YashBansod,

This is not CARLA related. You can choose which maps to include in a build in Unreal's Project Settings

[image]

YashBansod, 2018-03-02:
Even if I change the List of Maps to include to only the map I am using, The packaging process still includes all the assets. Is there a way to exclude the assets that are not used in the game?

YashBansod, 2018-03-02:
@nsubiron Even if I change the List of Maps to include to only the map I am using, The packaging process still includes all the assets. Is there a way to exclude the assets that are not used in the game?

nsubiron, 2018-03-05:
@YashBansod Maybe you can search in Unreal Engine's forums for this, e.g. https://answers.unrealengine.com/questions/184448/package-single-level.html

YashBansod, 2018-03-13:
I tried out all the methods available in the UE forums but none of them seemed to work.

But, I found that even though Carla project requires most of the assets in the project to be there in the project directory, I think it doesn't include those in the binaries if they are not needed.
I tried manually removing the asset folders from the project directory while the Carla project in UE4 Editor was open and saw that for some assets (like buildings) its fine to delete them from the project if not used, but deleting some assets (like Vegetation etc.) give error in the terminal UE4 was opened from (even if they are not used in the project). Nevertheless, I found that deleting the assets not required by my customised project didn't reduce the size at all. So, I think I can assume that while building the package of Carla, only required assets are built into the binary file.

closed 2018-03-13 by YashBansod