carla-simulator/carla #249: Are community pull requests allowed in Carla?

YashBansod, 2018-02-26:
Wanted to know if community pull requests are allowed for Carla.

I cloned the Carla master repository: `git clone https://github.com/carla-simulator/carla.git`
Created a branch: `git checkout -b branch_name`
Added the changes to the branch: `git add .`
Committed the changes: `git commit`
and push the changes to remote: `git push -u origin branch_name`

but I get the error: **unable to access 'https://github.com/carla-simulator/carla.git/': The requested URL returned error: 403**

Am I doing something wrong? or does Carla's git repo restrict this?

marcgpuig, 2018-02-26:
Hi @YashBansod,

Thanks for think on contributing into CARLA :)

First of all you must [fork](https://help.github.com/articles/about-forks/) our repository.
Then, please read the oficial github help on [Creating a pull request from a fork](https://help.github.com/articles/creating-a-pull-request-from-a-fork/).

It's easy!

YashBansod, 2018-02-27:
Thanks a lot @marcgpuig

closed 2018-02-27 by YashBansod