carla-simulator/carla #254: Modifying driving behaviour of Individual Non-player agents

ashishm-io, 2018-03-01:
I inted to modify driving behavior of individual non-player agents (by adding temporal noise). 

I tried modifying the file [WheeledVehicleAIController.cpp](https://github.com/carla-simulator/carla/blob/master/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/AI/WheeledVehicleAIController.cpp) but it modifies the behavior of all non-player agents simultaneously.

How do I modify the driving commands that a single non-player agent receives from the AI?

nsubiron, 2018-03-06:
Hi @ashishm-io,

Each vehicle gets its instance of `AWheeledVehicleAIController`, thus changes there affect all vehicles. Controllers "posses" the vehicle, so you can access your vehicle with `GetPossessedVehicle()`. In any case, you don't have enough info here to distinguish which car are you possessing.

Probably the best would be to create your own controller, spawn a vehicle (not with VehicleSpawner), and attach your controller to it.

ashishm-io, 2018-03-07:
Hi @nsubiron,

Thanks for your reply.
I am saving the address of individual possessed vehicle as a work around and attaching my controller to it.
Can you please point me to the code which passes individual vehicles iteratively to the AI Controller?

nsubiron, 2018-03-07:
@ashishm-io The controller is spawned right after spawning the vehicle by the vehicle itself
https://github.com/carla-simulator/carla/blob/957d44bb38ba1fb04041dd935cf3ac41f675121e/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Vehicle/VehicleSpawnerBase.cpp#L104-L107

ashishm-io, 2018-03-07:
@nsubiron thank you!

closed 2018-03-07 by ashishm-io

christianjcc, 2020-10-01:
Would anyone happen to know what the default CARLA non-player/vehicle driver model based on such as Intelligent Driver Model (IDM) or Optimal Velocity Model (OVM), etc? Could not find it described anywhere.