carla-simulator/carla #258: Pedestrian, what is the algorithm to control pedestrain's path.

5guo, 2018-03-02:
Firstly, the code of carla-simulator/carla is so beautiful. I have a problem when I read it, where can I find the program about control pedestrain? When a pedestrain is generated with position and speed, then, how to calculate his next position?

nsubiron, 2018-03-02:
Hi @5guo,

Pedestrian are using the built-in navigation system that comes with Unreal Engine. The trick is to add a `UCrowdFollowingComponent` to our `AWalkerAIController` class. Pedestrians avoid each other and find their way following the navmesh present in the maps, we just tell them where to go and their max speed. The map also has areas with different weights, so pedestrians prefer to walk on the sidewalk and cross the road only when necessary.

Just mention that this system has some issues, as in #6 or #7.

5guo, 2018-03-05:
Thank you so much for your attention and answer. The pedestrians can find their way in the maps and avoid each other, all of these are provided by Unreal Engine's algorithm? So, where can I find the algorithm's detail?

nsubiron, 2018-03-05:
The navigation mesh is generated by adding a [navmesh volume](https://docs.unrealengine.com/latest/INT/Engine/AI/BehaviorTrees/QuickStart/2/) to the scene. Then the paths are calculated using an A* algorithm. You can find more details in Unreal Engine documentation or even their source code.

5guo, 2018-03-06:
All pedestrians can avoid collision from each other, was this function provided by UE4's AI Modules? Or It was implemented with Carla's algorithm?

nsubiron, 2018-03-06:
All the pedestrian movement is done by Unreal Engine's methods, including avoidance.

The only thing we implement is a basic detection of vehicles, when a vehicle appears to be in the pedestrian route the pedestrian momentarily pauses its movement. We implement this to avoid #7 as much as possible.

simmates, 2018-03-06:
It would be nice to be able to implement an interface to manage the control of pedestrians - especially to generate corner cases.

nsubiron, 2018-03-14:
Controlling all the agents in the scene is a feature we have in our backlog, follow the progress at #184.

closed 2018-03-14 by nsubiron

slyer0009, 2019-02-11:
> Hi @5guo,
>
> Pedestrian are using the built-in navigation system that comes with Unreal Engine. The trick is to add a `UCrowdFollowingComponent` to our `AWalkerAIController` class. Pedestrians avoid [...]

Can you explain how to add a UCrowdFollowingComponent to get this working? I'm unable to get my spawned pedestrians to move using the navmeshs in Town01
Thanks!

jingluoquanergy, 2021-06-18:
@nsubiron I've observed pedestrians intermittently walk though vehicles for CARLA with commit id  5ce501d079981d931554b6bb0a9e39ed71d423a2. It tends to happen when the vehicles are at slow speed(e.g. making a turn).