carla-simulator/carla #261: Control Non-Player Agents

ajax98, 2018-03-04:
I have peered through the CARLA code and there is a way to control the current player. Is there a way to control the position of non-player agents?

YashBansod, 2018-03-04:
@ajax98 making the NPCs user controllable is already a feature request for carla (see issue #184).
But if by control the position of NPCs you just mean the position where they spawn, that is already doable.

closed 2018-03-05 by nsubiron

Marwa215, 2018-06-18:
Hi, I have the same issue my phd project about automatic scenario generation by automatically creating a set of events like providing dynamic objects (weather conditions, traffic signs, cars, pedestrians....etc) based on the trainee performance level...so I need to control adding these objects...could I know when these features will be released.
Thanks 
Marwa