carla-simulator/carla #262: Issue with the map data created for new maps in Carla.

YashBansod, 2018-03-04:
I created a bare bones map in the Carla Project and It has only the road, landscape and a player vehicle. I packaged it for Linux. Then I started the newly created map in the server mode with the following options:
`/Game/Maps/TownA -carla-server -windowed -ResX=800 -ResY=600 -benchmark -fps=25`

Here **TownA** is the newly created map.
I copied the **TownA.png** and the **TownA.txt** to **PythonClient > carla > planner**.

The Image of TownA.png that was created
[image]

The TownA.txt just has these contents:
> Map name = TownA
> Size = 1,560x810 pixels
> Density = 54.767 cm/pixel
> World-To-Map Transform (T|R|S) = (544.000000,-10748.000000,-22.000000|0.000000,0.000000,0.000000|1.000000,1.000000,1.000000)
> Map Offset = (X=-1643.022 Y=-1643.022 Z=0.000)

[TownA.txt](https://github.com/carla-simulator/carla/files/1778723/TownA.txt)

Then I ran the **manual_control.py** with the following arguments:
`manual_control.py -v -m=TownA`

But, I get the following errors:

> DEBUG: (localhost:2000) connected
> DEBUG: (localhost:2000) disconnecting
> Traceback (most recent call last):
>   File "/home/uidk4283/PycharmProjects/CARLA_Client_0.8.0/manual_control.py", line 413, in <module>
>     main()
>   File "/home/uidk4283/PycharmProjects/CARLA_Client_0.8.0/manual_control.py", line 401, in main
>     game = CarlaGame(client, args.lidar, args.map_name)
>   File "/home/uidk4283/PycharmProjects/CARLA_Client_0.8.0/manual_control.py", line 144, in __init__
>     self._map = CarlaMap(city_name, 16.43, 50.0) if city_name is not None else None
>   File "/home/uidk4283/PycharmProjects/CARLA_Client_0.8.0/carla/planner/map.py", line 44, in __init__
>     self._graph = Graph(city_file, node_density)
>   File "/home/uidk4283/PycharmProjects/CARLA_Client_0.8.0/carla/planner/graph.py", line 50, in __init__
>     self._resolution = string_to_node(linegraphres)
>   File "/home/uidk4283/PycharmProjects/CARLA_Client_0.8.0/carla/planner/graph.py", line 13, in string_to_node
>     return (int(vec[0]), int(vec[1]))
> ValueError: invalid literal for int() with base 10: 'Map Offset = (X=-1643.022 Y=-1643.022 Z=0.000)\n'
> 
> Process finished with exit code 1

I have few observations:

1. Contents of TownA.txt not written in the same format as its in Town01.txt
2. TownA.png is a lane image. (Similar to Town01Lanes.png)
3. Map of TownA.png was not generated (The one like Town01.png) rather Lane image was generated with TownA.png name.
4. TownACentral.png (the one that should be like Town01Central.png) was not generated.

Can someone tell me a solution for resolving the error I am getting.

For convenience I am attaching the link to the Carla Map that I created in UE4:
https://learnermanipal-my.sharepoint.com/:u:/g/personal/yash_bansod_learner_manipal_edu/ERMlyD2EC1FEp-ed-GS6cFkB6o2WYcaukJ4AfwZIJHXDwg?e=CJ335X

ashishm-io, 2018-03-14:
I'm facing a similar issue when trying to run a custom-made map as a server.
Is there a specific process to follow while packaging a custom map?

umrichen, 2018-05-10:
I'm having exactly the same issue here. How to generate the correct .pngs and txts in the same format as those given with Town01/02?
Thanks!

felipecode, 2018-05-16:
Hello @YashBansod ,
First, I am sorry for taking too long to answer the issue. I was in a deadline period.
Very nice work you made with the map!
Unfortunately CARLA does not  have a good system for incorporating a planner to newly created maps,
however I will try to help you as much as possible. We are currently working hard in a general solution for that, i.e having a proper city description format to plan with.

The first thing you need is to generate the maps. You were able to already generate
that lane maps ! That is great. 
With this you will need have the display map, TownA.png. Your solution is good you could use any map, with the same size as the lane map, for the display map. (Is the map that is shown on manual_control.py for instance).
The central lanes is not really needed for the master branched I was using that for the checkpoint generation. 

What you are missing is the graph description file ( TownA.txt) 
that defines all the conections and nodes. Here is an example, used for Town02.
I will explain line by line:

> 5.4400,-107.48000,-0.22000000
> 0.000000,0.000000,0.000000
> 1.000000,1.000000,1.000000
> -16.43022,-16.43022,0.000
> 25, 25
> 0,10 0,24 14
> 0,24 0,10 14
> 24,24 6,24 18
> 6,24 24,24 18
> 24,0 24,10 10
> 24,10 24,0 10
> 0,0 24,0 24
> 24,0 0,0 24
> 0,10 0,0 10
> 0,0 0,10 10
> 24,10 24,16 6
> 24,16 24,10 6
> 0,10 6,10 6
> 6,10 0,10 6
> 6,24 0,24 6
> 0,24 6,24 6
> 6,10 17,10 11
> 17,10 6,10 11
> 6,24 6,16 8
> 6,16 6,24 8
> 24,16 24,24 8
> 24,24 24,16 8
> 6,16 6,10 6
> 6,10 6,16 6
> 24,16 17,16 7
> 17,16 24,16 7
> 17,16 6,16 11
> 6,16 17,16 11
> 17,10 24,10 7
> 24,10 17,10 7
> 17,16 17,10 6
> 17,10 17,16 6

Line 1: 
5.4400,-107.48000,-0.22000000

Is the position of (0,0,0) in world coordinates. Basically where the origin was placed in unreal.

Line 2:
0.000000,0.000000,0.000000
Is the rotation with respect to the origin, in this case zero

Line 3: Scale , not used
Line 4:
-16.43022,-16.43022,0.000
Is the transformation you need to do to move the origin to the upper right point of the image
Line 5:
25, 25 The graph size 
From line 5 to the end:
> 0,10 0,24 14  

The graph nodes and conected
This one for instance is a node in position 0,10, a node in position 0,24, with a distance of 14 between them. All of this is in graph node units.
There is also somethings hardcoded for planner configuration:
If you look at planner/city_track.py lines 18/19
        self._node_density = 50.0
        self._pixel_density = 0.1643
These variables determine, respectively, the number of pixels for each graph node
and the number of pixels per meter.

Hope I was able to help a bit. We are very interested in the results you get ! 
We will keep you guys updated on the development of the new planning system.

Cheers.

YashBansod, 2018-05-17:
Firstly, thanks a lot for the detailed explaination @felipecode .
With regard to your post above, I have a few doubts:

> Map name = TownA
> Size = 3,060x3,060 pixels
> Density = 54.767 cm/pixel
> World-To-Map Transform (T|R|S) = (-2000.000000,-2000.000000,-22.000000|0.000000,0.000000,0.000000|1.000000,1.000000,1.000000)
> Map Offset = (X=-1643.022 Y=-1643.022 Z=0.000)

These are the contents of my TownA.txt (Autogenerated)
This I will need to change to: 

> -2000.000000,-2000.000000,-22.000000
> 0.000000,0.000000,0.000000
> 1.000000,1.000000,1.000000
> -16.43022, -16.43022, 0.000

But I don't understand wht you mean by graph size (the value 25, 25).
[image]

Do you mean Map size? The Map Size X and Y for Town01 was 25, 25 respectively. 
But for my map its 200, 200
so I put that on line 5:
> 200, 200

After this from line 5 to end, How do I get those graph node values and the distance between them from?

The auto-generated txt file for TownA (One I created) shows
> Density = 54.767 cm/pixel

So do I need to do some changes in :
> If you look at planner/city_track.py lines 18/19
> self._node_density = 50.0
> self._pixel_density = 0.1643
> These variables determine, respectively, the number of pixels for each graph node

felipecode, 2018-05-17:
I think you did everything correct !
The pixel density for your case should be 0.54767 m/pixel
Your map is indeed 200,200.
I think that the node density should be the same.
Again, we have no simple way of generating a graph. We are sorry for that. 
What I did, long time ago, was to just print the built graph when 
building the map.

Basically the graph is generated on 

https://github.com/carla-simulator/carla/blob/c7093b58ad98b4558a8d9a5768f2727cad5df069/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/MapGen/GraphGenerator.cpp#L96-L113

The graph is a double connected list. You could go inside the structure and print it.
Maybe you would need to do a simple conversion script to get the graph in the format I described in the above post.
The doublyconnected list is defined here:
https://github.com/carla-simulator/carla/blob/master/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/MapGen/DoublyConnectedEdgeList.cpp

closed 2018-06-07 by nsubiron