carla-simulator/carla #264: Bounding box of player vehicle Mustang

n-akram, 2018-03-05:
Hi,

I wanted to know the car parameters (length, width, height) of player cars. Player measurement does not include bounding_box. I read that the default vehicle is Mustang and I should look for these in Content/Blueprints/Vehicles. But i was not able to locate this. Is it released for public already?

#135

YashBansod, 2018-03-05:
@n-akram even though the player measurement doesn't include the bounding_box information, you still get it as a part of npc measurements. 
Just try to dump the measurements with 0 npcs in the scene and you will see what I mean.
Also, I can confirm that the first vehicle instance in the measurements is that of the player vehicle.

nsubiron, 2018-03-06:
@n-akram Yes, it is publicly released. You can see the bounds opening the Mustang blueprint

[image]

Note than although the height is not currently adjusted we plan to fix this in the coming release (#83).

nsubiron, 2018-03-06:
@YashBansod 

> Also, I can confirm that the first vehicle instance in the npc measurements is always of the player vehicle.

This is arbitrary and you should not count with this. Actually, we have noticed this issue and we are going to fix it in the next release; we plan to add the box extend to the player measurements and remove the player from the non-player agents list.

n-akram, 2018-03-09:
Hi, 

Thank you for your inputs. I have tried getting the bounding box. However, I was not able to locate the player vehicle information successfully in the npc measurements. I have used the method of comparing locations to find player in npc. Sample code is as `below.

`measurements, sensor_data = self.client.read_data()
player_loc = measurements.player_measurements.transform.location
for agent in measurements.non_player_agents:
    if(agent.HasField('vehicle')):
          if((agent.vehicle.transform.location.x - player_loc.x) and (agent.vehicle.transform.location.y - player_loc.y) and (agent.vehicle.transform.location.z - player_loc.z)):
               return(agent.id)`

I have also tried printing the location information, for all agents. But I could not find any. This is crucial for me as I do not want to include player vehicle in npc vehicles in my studies.

Please help me out with this, or point me in right direction.

I appreciate your help. 

Thanks and regards,
-Akram

nsubiron, 2018-03-09:
Hi @n-akram,

The code is already changed to remove the player from the npc list. It is not yet in master but awaiting approval at #272. If urgent you can building from source the branch `issue#268`, otherwise you can wait for the 0.8.0 release that will be ready hopefully next week.

closed 2018-03-22 by nsubiron