carla-simulator/carla #267: Blank screen when building carla project.

SpencerLo-CMU, 2018-03-06:
I follow the installation stpes in official guide. 

When I run "./Rebuild.sh" to build carla, it comes up with a blank screen. 

At this time, I type "nvidia-smi" to check GPU status, it shows "Unable to determine the device handle for GPU 0000:02:00.0: GPU is lost." I have no other choice but to reboot my machine.

I also try the binary version by inputting "./CarlaUE4.sh". It goes to blank screen immediately.

I am using:
- Ubuntu 16.04
- GTX 1080 with driver 390.X

Any idea on this problem?

juaxix, 2018-03-06:
Hi @SpencerLo-CMU.
Your problem is related with your card configuration more than with the use of CARLA. 
If you search in the nvidia forums you will find your issue in some of their posts, where they say it is often connected to overheating or failure of power source. 
Try to use the experimental branch [Optimizations](https://github.com/carla-simulator/carla/tree/Optimizations) and set the default quality settings to Low. Also, set the unreal quality settings to Low.
Hope that reduce the gpu use a little bit.
Thanks.

SpencerLo-CMU, 2018-03-06:
Thanks for your quick replies! @juaxix 

2 more questions follow up:
- What is the main difference between the master branch and Optimizations branch.
- I am a newbie to carla and UE4. Please tell me where I can change the default quality settings you mentioned above.

YashBansod, 2018-03-07:
@SpencerLo-CMU, any updates done to carla are added to the master branch only after its compatibility and stability on various systems have been verified. All branches are reviewed before merging to the master branch.
As @juaxix said the Optimiztions branch is an experimental branch. (maybe where they are trying to set up various optimizations for carla to use the compute resources more efficiently).
If you go to the git page of carla, you can see many other branches along with master. These branches are basically the work towards various feature requests and bug fixes to the master branch.

In UE4 you can change the default quality settings from the settings option in the toolbar of UE4 editor, for Carla I don't think that functionality is available in the main branch

SpencerLo-CMU, 2018-03-07:
Gotcha! Thanks for detailed response! @YashBansod

closed 2018-03-07 by SpencerLo-CMU