carla-simulator/carla #269: How to control gear shift in Carla?

YashBansod, 2018-03-06:
I was trying to implement a basic velocity control algorithm using PID, in carla. 
But I notice that the car keeps shifting the gears between kmph to 40kmph (1st and 2nd gear) and 50-60kmph (3rd gear). This is causing a fluctuating speed and control commands from my PID controller.

**Is there a way to control the gear shift or outright disable it?**

The following plots may help visualize the problem better.All plots are with the same tuning parameters:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

nsubiron, 2018-03-07:
> Is there a way to control the gear shift or outright disable it?

We don't have currently any way of shifting gears from the client side, it's just not added to the vehicle control. I'm actually surprised no one asked for this up until now :D I think is reasonable to add this to our backlog.

Meanwhile you can tune the vehicle's engine model inside Unreal Editor, would it work adding a single very long gear?

[image]

YashBansod, 2018-03-07:
Thanks @nsubiron. 
I think setting a very long gear may solve the purpose for now.

closed 2018-03-07 by YashBansod

fbuchinger, 2018-04-14:
+1 for the manual gear shift, would love to use my logitech g25 shifter for that.

AlienTech83, 2018-05-11:
+1 for this, would really be good to be able to shift gears using manual control (either from keyboard or from a steering wheel)

NYJiaxing, 2018-08-08:
@YashBansod  
Hello Sir,

Would you mind if you can share with me how you get the vehicle's speed data? And did you find where is the vehicle's yaw angle data?

Thank you so much

Derekabc, 2019-08-13:
Hello, I am not sure what is the meaning of " setting a very long gear ", does that mean we can set munual_gear_shift = 10 (big number?). thank you very much!!