carla-simulator/carla #276: Strange effects in Camera Images

simmates, 2018-03-09:
I'm experiencing some strange effects that are happening when I connect to Carla via the python client that I want to describe here.
First of all it becomes quite apparent that whenever the car is moving the lighting seems very unnatural - e.g. there are a lot of reflectons in sunny weather conditions - especially a lot of short flashes on reflective surfaces that repeat. I give an example:
[image]
[image]
[image]
[image]
These are only 4 of 5 consecutive frames but on the fence you see a lot of flashes that happen almost every frame during the whole time the car approaches the intersection and even when it is still very far away you can see some of them.

Another example is the shadow on the guardrail in these frames (there should be no shadow but a shadow seems to be moving on the guardrail - this happens repeatedly with nothing casting a shadow there).
[image]
[image]
[image]
[image]

It appears that when I start an instance in the Unreal editor these strange effects are not present... (maybe a different camera is used there?).

TheNihilisticRobot, 2018-03-12:
Hi @simmates
 
The editor camera and the capture camera used in the built game are indeed different and don't work the same They give slightly different results in various aspects, including the excessive flares you are reporting about. We expect to make them practically equal but we still have not a proximate date of getting this as we need a lot of information from the Epic team.

As for the Issue with the shadows on the guardrail my only guess is that the lower quality shadow rendered when the camera is far away is calculating the percentage of shadow projected unto the model and shading it uniformly. Since the model has a lot of concave surfaces, this average lighting is pretty dark.

simmates, 2018-03-20:
i tried getting rid of the excessive flare by changing rendering parameters as well as material parameters but so far I have no satisfactory results. I'm wondering if anybody else finds this disturbing as it seems very artificial. If anyone had tips on how to improve the settings maybe more people can benefit from it.
Also I find that trees and sometimes shadows just appear and textures that aren't very far away change abruptly - I suppose it has something to do with clipping distances etc. but changing parameters in the world settings or near clipping planes doesn't help. Any suggestions?

TheNihilisticRobot, 2018-03-20:
We've been encountering the same problems as you with the flares in the capture, unfortunately its not something we have a fix for, yet.

The other things you mention are fixable through some console commands, though, "fixing" them would have negative impact on the performance, since they are the product of measures to cheapen the render costs:

The popping in the vegetation and other models is caused by the Level of Detail. We have various versions of the same model with different polygonal levels to replace the final model depending on the distance from the camera. Ideally, these changes would be imperceptible but more tweaking would be needed to reach that ideal point. You can force the highest LoD on everything by typing "r.forcelod 0" in the Unreal Engine console.

The texture changes are probably caused by a similar feature but for the textures: MipMapping. To disable this type "r.TextureStreaming" in the Unreal Engine console.

simmates, 2018-03-20:
It is possible to record the ingame camera with a matinee - do you know how to change its position? The settings in the settings file only affect the SceneCapture Camera used by the python client right?

TheNihilisticRobot, 2018-03-21:
You may use Matinee or Cinematics if their functions meet your needs, however using them you'll only be able to render one camera at a time and no ground truth whatsoever. You'll have to work from the unreal editor where you'll be able to configure almost anything you need. 

You can't change the position of the player camera unless you manually change our C++ code (Though you may manually move it during gameplay with the arrow keys). The configuration files, as you've said, works only with the SceneCapture camera.

closed 2018-05-24 by TheNihilisticRobot