carla-simulator/carla #278: Building/running Carla on Centos  - Help and ifnormation requested.

gilamid, 2018-03-12:
I would appreciate any help with respect to building or running carla on CentOS. Any information on version numbers, libraries, compile , etc... . I have see a succesful report from yiakwy  ( dated december 2017 ) and ability to do that, but no details were available. 

Any helps that me save me time will help !
Thanks !

marcgpuig, 2018-03-12:
Hi @gilamid,

You need to compile Unreal and CARLA with the same compiler, we use:
- Clang 3.9

And the same for all his dependencies:
- Libc++ 3.9
- Boost 1.64
- Protobuf 3.3.0

Boost and Protobuf must be linked with libc++.
Please check out the [`Setup.sh`](https://github.com/carla-simulator/carla/blob/master/Setup.sh)

Relative to #103, maybe we can ask to @yiakwy and see what he has discovered.
If he can share something in this issue, it will be incredibly helpful for the community ;)

closed 2018-04-03 by marcgpuig

qhaas, 2020-06-24:
I'm trying to build only the CARLA 0.9.7 PythonAPI client in EL7 / Python3.  I've gotten it to compile with the SCL's llvm-toolset-7.0's clang 7 in CentOS7, but I keep getting this error when I try to `import carla`:

```
>>> import carla
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/nqh/.local/lib/python3.6/site-packages/carla/__init__.py", line 8, in <module>
    from .libcarla import *
ImportError: /home/nqh/.local/lib/python3.6/site-packages/carla/libcarla.cpython-36m-x86_64-linux-gnu.so: undefined symbol: _ZTVN10__cxxabiv117__class_type_infoE
```

From reading the build scripts, it seems that libstdc++ should be used for compiling the libcarla client and libc++ the server, so I'm not sure how libc++ is getting introduced.