carla-simulator/carla #279: Running Optimizations Branch on Headless Server

bhairavmehta95, 2018-03-12:
Is it possible to compile the `optimizations` branch on a headless server? I am getting errors similar to the ones others have had on here, but the solutions seemed to be centered around opening the UE4Editor and linking materials, which probably can't be done on a server (unless someone else has done this)? My server doesn't seem to launch, since after running `sudo lsof -i :2000` the port seems to be unused.

```
1001[2018.03.12-21.19.38:733][  0]LogLinker: Warning: Can't find file for asset '/Game/Static/Buildings/ResidentialTallWStairs/N2/_ResidentialTallWStairs_GroundFloor_Door_Mat_N2_OcclusionRoug
hnessMetallic' while loading ../../../../optimization-carla/Unreal/CarlaUE4/Content/Static/Buildings/ResidentialTallWStairs/N2/ResidentialTallWStairs_Door_N2_Inst.uasset.
[2018.03.12-21.19.38:733][  0]LoadErrors: Error: /Game/Static/Buildings/ResidentialTallWStairs/N2/ResidentialTallWStairs_Door_N2_Inst : Can't find file for asset. /Game/Static/Buildings/Resid
entialTallWStairs/N2/_ResidentialTallWStairs_GroundFloor_Door_Mat_N2_OcclusionRoughnessMetallic
1001[2018.03.12-21.19.52:456][  0]LogLinker: Warning: Can't find file '/Game/Static/Buildings/ResidentialTallWStairs/N2/_ResidentialTallWStairs_GroundFloor_Door_Mat_N2_OcclusionRoughnessMetal
lic'
[2018.03.12-21.19.52:456][  0]LoadErrors: Failed to load /Game/Static/Buildings/ResidentialTallWStairs/N2/_ResidentialTallWStairs_GroundFloor_Door_Mat_N2_OcclusionRoughnessMetallic._Residenti
alTallWStairs_GroundFloor_Door_Mat_N2_OcclusionRoughnessMetallic Referenced by ResidentialTallWStairs_Door_N2_Inst
1001[2018.03.12-21.19.52:457][  0]LogLinker: Warning: Can't find file for asset '/Game/Static/Buildings/ResidentialTallWStairs/N2/_ResidentialTallWStairs_GroundFloor_Door_Mat_N2_OcclusionRoug
hnessMetallic' while loading NULL.
[2018.03.12-21.19.52:457][  0]LoadErrors: Failed to load /Game/Static/Buildings/ResidentialTallWStairs/N2/_ResidentialTallWStairs_GroundFloor_Door_Mat_N2_OcclusionRoughnessMetallic._Residenti
alTallWStairs_GroundFloor_Door_Mat_N2_OcclusionRoughnessMetallic Referenced by ResidentialTallWStairs_Door_N2_Inst
```

juaxix, 2018-03-14:
Yes @bhairavmehta95 you can, the errors are related to the assets, we are preparing the files at this moment, so, please, give us a couple of days and then you will be able to download them from the [Update](https://github.com/carla-simulator/carla/blob/master/Update.sh) script ([more info](http://carla.readthedocs.io/en/latest/how_to_build_on_linux/#updating-carla)); anyway, try [this](https://drive.google.com/open?id=1xEO2jxiQV43mwoeZdy39VglzvlL_l56B) meanwhile.
Kind regards

bhairavmehta95, 2018-03-14:
I've added the newest content files, so that fixed the errors issue, but the server still seems to hang

```
[2018.03.14-17.45.40:863][  0]LogContentBrowser: Native class hierarchy updated for 'HierarchicalLODOutliner' in 0.0010 seconds. Added 1 classes and 2 folders.
[2018.03.14-17.45.40:863][  0]LogLoad: (Engine Initialization) Total time: 122.75 seconds
[2018.03.14-17.45.40:863][  0]LogLoad: (Engine Initialization) Total Blueprint compile time: 0.00 seconds
[2018.03.14-17.45.40:863][  0]LogExternalProfiler: No external profilers were discovered.  External profiling features will not be available.
[2018.03.14-17.45.41:199][  0]LogSlate: Took 0.000766 seconds to synchronously load lazily loaded font '../../../Engine/Content/Slate/Fonts/Roboto-Regular.ttf' (155K)
[2018.03.14-17.45.41:205][  0]LogSlate: Took 0.000155 seconds to synchronously load lazily loaded font '../../../Engine/Content/Slate/Fonts/Roboto-Bold.ttf' (160K)
[2018.03.14-17.45.41:219][  0]LogSlate: Took 0.000140 seconds to synchronously load lazily loaded font '../../../Engine/Content/Editor/Slate/Fonts/FontAwesome.ttf' (139K)
[2018.03.14-17.45.41:437][  0]LogContentStreaming: Texture pool size now 2000 MB
[2018.03.14-17.45.41:469][  0]LogRenderer: Reallocating scene render targets to support 804x360 Format 10 NumSamples 1 (Frame:1).
[2018.03.14-17.45.41:469][  0]LogSlate: Took 0.000107 seconds to synchronously load lazily loaded font '../../../Engine/Content/Slate/Fonts/Roboto-Light.ttf' (159K)
[2018.03.14-17.45.41:785][  1]LogLinux: Setting swap interval to 'Immediate'
[2018.03.14-17.45.41:785][  1]LogLinux: Warning: Unable to set desired swap interval 'Immediate'
[2018.03.14-17.45.41:845][  1]LogLinux: Setting swap interval to 'Immediate'
[2018.03.14-17.45.41:845][  1]LogLinux: Warning: Unable to set desired swap interval 'Immediate'
[2018.03.14-17.45.43:463][ 26]LogAssetRegistry: Asset discovery search completed in 113.0123 seconds
[2018.03.14-17.45.43:492][ 26]LogCollectionManager: Rebuilt the object cache for 0 collections in 0.000001 seconds (found 0 objects)
[2018.03.14-17.45.43:492][ 26]LogCollectionManager: Fixed up redirectors for 0 collections in 0.000116 seconds (updated 0 objects)
```

After this section, I get nothing. I've made sure to build each part from scratch with the Optimizations (i.e Unreal, Carla, UE4Editor, etc) with the provided scripts, but after this section, nothing else shows up.

Can't seem to connect with the Python client_example, but for some reason, when I run the cpp `make test`, all the tests seem to pass (all of the Python examples can't seem to connect to the server, and since the world port isn't in use after checking with `sudo lsof -i :world_port_used`, I'm not sure what the issue is.)

juaxix, 2018-03-14:
First, put the Scale mode in Low quality for testing first.
Copy CarlaSettings.ini from the [Docs/Example.CarlaSettings.ini](https://github.com/carla-simulator/carla/blob/Optimizations/Docs/Example.CarlaSettings.ini) into [Unreal/CarlaUE4/Config](https://github.com/carla-simulator/carla/tree/Optimizations/Unreal/CarlaUE4/Config) and set 

> QualityLevel=Low

then run a client from [carla/PythonClient/](https://github.com/carla-simulator/carla/tree/Optimizations/PythonClient), like [python client_example.py](https://github.com/carla-simulator/carla/blob/Optimizations/PythonClient/client_example.py)
or other ,depends on what you want.

Finally, hit Play button to Run the server that will connect with the client.

bhairavmehta95, 2018-03-14:
Is there any way to do this without a display?

juaxix, 2018-03-15:
Yes, you have to build it, using the [Package.sh](https://github.com/carla-simulator/carla/blob/Optimizations/Package.sh) once you configure the project and scripts or download the compiled project from the repo in [releases](https://github.com/carla-simulator/carla/releases/) section.

bhairavmehta95, 2018-03-20:
Thank you @juaxix !

For any one who's looking at this in the future, be sure to pay attention to the last line of output from the script, which tells you where the resulting binary is.

closed 2018-03-20 by bhairavmehta95