carla-simulator/carla #280: Teleport the vehicle

YangZhang4065, 2018-03-13:
Hi,

From what I have read, I understand that the manual car control is implemented. Instead of manually control the vehicle, is there a way to teleport the player vehicle to a specific location via python client? Thank you.

YashBansod, 2018-03-13:
Can you explain your application? How is teleportation related to any of the applications of Carla?
 
There is a way to spawn the vehicle to a location, but this is only at the start of the episode.

YangZhang4065, 2018-03-13:
Sorry. I am not working on the autonomous driving, instead, I am testing my semantic segmentation network on a particular object (in this case it is the player's vehicle) in the virtual environment. So it would be appreciated if I can spawn/teleport the vehicle to the different locations with some diverse backgrounds to collect as much training/testing data as possible within one episode.

YashBansod, 2018-03-13:
Don't know if teleporting while an episode is ongoing is possible, but why not just capture 10000 frames while the car runs in autopilot mode and then just randomly sample from this dataset. In a way it is doing the same thing.

nsubiron, 2018-03-13:
Hi @YangZhang4065,

Teleporting the vehicle is not currently possible. I'm going to put this in our backlog to have it present, but I'm not really sure we will be able to do this. It has some difficulties like changing the server-client protocol, or making sure there is no collision on the target position, that make it non-trivial; so we'll need to study if this is really useful before doing those changes.

As a workaround, what @YashBansod said might be an option.

In any case thanks for the feedback :)

YangZhang4065, 2018-03-13:
Hi @YashBansod @nsubiron ,

I see. I understand it would be a difficult task to detect the potential colission. The workaround sounds great and I will try it first. Thank you:)

nsubiron, 2018-08-01:
This is already implemented in API 0.9.0.

closed 2018-08-01 by nsubiron