carla-simulator/carla #284: Help requested: Runnign carla  on brand new headless UBUNTU  server

gilamid, 2018-03-15:
Just installed a brand new UBUNTU 16.04 server, headless, but with an NVIDIA GTX 1050Ti GPU 
Trying to run CARLA using a remote sessions ( ssh -X <new-server) ) - carla is crashing. 
It looks like I am missing some packages in the installation, but I can not figure out which
any help will be appreciated.

These are the issues I see in the log file: 

LogPluginManager: Mounting plugin Carla
LogFileManager: Warning: ReadFile failed: Count=0 Length=4096 Error=errno=21 (Is a directory) for file
Init: Using libcurl 7.48.0-DEV

And then:
2018.03.15-11.49.52:907][  0]LogInit: Initializing SDL.
[2018.03.15-11.49.52:907][  0]LogInit: Using SDL_WINDOW_OPENGL
[2018.03.15-11.49.52:997][  0]LogInit: Warning: Could not initialize SDL: XRandR support is required but not available
[2018.03.15-11.49.52:997][  0]LogInit: Warning: FDisplayMetrics::GetDisplayMetrics: PlatformInitMultimedia() failed, cannot get display metrics
[2018.03.15-11.49.52:997][  0]LogLinux: Selected Device Profile: [LinuxNoEditor]

and then , the grand finale:

LinuxApplication::CreateLinuxApplication() : PlatformInitMultimedia() failed, cannot create application instance.
[2018.03.15-11.47.54:543][  0]LogLinux: Error: appError called: Assertion failed: Assertion failed:  [File:/home/nsubiron/UnrealEngine_4.17/Engine/Source/Runtime/Core/Private/Linux/LinuxApplication.cpp] [Line: 36] 
FLinuxApplication::CreateLinuxApplication() : PlatformInitMultimedia() failed, cannot create application instance.

Signal 11 caught.

nsubiron, 2018-03-15:
Hi @gilamid,

Can you do the ssh connect without the `-X` flag? That causes Unreal to detect the X display and it tries to render to it usually crashing.

nsubiron, 2018-03-15:
@felipecode Please add this to your tutorial to run headless as it's a common issue

gilamid, 2018-03-15:
Thanks, indeed without the -X , the server runs with no warnings. This bring up a different question: you mention that it crashes when it tries to render to the Xdisplay.  what is the way to redirect the rendering to a remote display ?

nsubiron, 2018-03-15:
I am not aware of any way of redirecting to a remote display. Maybe with VirtualGL and TurboVNC you can achieve something like that, but we have never tried it.

ciumonk, 2018-03-24:
x11vnc perhaps?

closed 2020-03-19 by germanros1987