carla-simulator/carla #286: How to disable rendering in CARLA server?

GokulNC, 2018-03-20:
So when the CARLA server is run by using the script `CarlaUE4.sh` to launch it, is there any way to disable the rendering of the game window?

I use a camera in the simulator to get the current FoV from that camera. So just reading the image by `CarlaClient.read_data()` is enough for me to process things. I feel that I can increase my [FPS](https://github.com/carla-simulator/carla/issues/10) by reducing the processing overhead to **render** the game using Unreal Engine.

I tried using the flag `-carla-no-hud` as suggested in the [docs](http://carla.readthedocs.io/en/latest/how_to_run/), but still the rendering is taking place. 

So is there anyway to do this? Or it's not just possible or it doesn't matter not to render?
Thanks :)

marcgpuig, 2018-03-20:
Hi @GokulNC,

First of all `-carla-no-hud` disables the left-bottom text output from the server:

[image]

Currently is no possible to disable the render and I have the feeling that it will doesn't matter anyways, because regardless you see or not the render, it must be done anyway.

GokulNC, 2018-03-20:
Thanks @marcgpuig .

Eventhough the computations need to occur, I feel rendering it at a certain FPS (say 15) is quite resource-intensive.

Though there is no way to disable rendering as of now, can this issue be considered as a feature-request?

I will close this issue if this feature is implemented, or if there's an strong assertion that disabling rendering doesn't matter much.

Thanks :)

marcgpuig, 2018-03-20:
Please try adding these commands:

    -windowed -ResX=8 -ResY=8

So it will run on a `8x8` pixels window. It's a workaround but I hope you can win some extra fps. 😉 

Another solution is work without an actual screen, see our [documentation](http://carla.readthedocs.io/en/latest/carla_headless/).

For sure I'm going to add the *feature-request* tag because it is something that can improve the performance.
But we are currently working on other more critical optimizations, so do not expect big focus on this one.

Since it's more an *Unreal Engine issue*, I'm adding the *good first issue* too, so the community can make a good-and-easy contribution 😄 

Regards!

simmates, 2018-03-20:
Is it also possible to disable the HUD in the Editor?

simmates, 2018-03-20:
Ok it is possible in the console with showHUD 0

juaxix, 2018-04-13:
You can also set 

> r.ScreenPercentage 0

using the unreal console

marcgpuig, 2018-05-31:
Is this topic solved?

juaxix, 2018-05-31:
@marcpuig I think we should add a command for this, to disable the rendering in the viewport like this

ULocalPlayer* player = GEngine->GetGamePlayer(GetWorld(), 0);
player->ViewportClient->bDisableWorldRendering = true;

marcgpuig, 2018-05-31:
@juaxix 
Yeah looks good to me!

GokulNC, 2018-05-31:
@juaxix Yeah, that might help IDK :)

But as an end user like me, it would be better if there's a command-line flag or an option in CarlaSetting.ini or something like that to disable it easily, without recompiling from scratch.

Thanks :)

juaxix, 2018-06-01:
Ok, the feature has been implemented in the branch [issue-286](https://github.com/carla-simulator/carla/tree/issue%23286), you can wait for it to be merged into main and it will be added in CARLA 0.8.3. 
[image]

In CarlaSettings.ini, the new flag is inside the [Server] section, and it's called: DisableRendering, by default is false, Ah, it will hide the hud too if set to true. And the new command for the executable is called "-disable-rendering".

Cheers.

closed 2018-08-10 by stale[bot]

Johnnyswong, 2018-12-09:
I have added the "-disable-rendering" to arguments when I running the Carla.sh script, but it didn't worked. But changing the "DisableRendering" to true in the Carla.ini file worked.

aayushARM, 2019-03-14:
> I have added the "-disable-rendering" to arguments when I running the Carla.sh script, but it didn't worked. However changing the "DisableRendering" to true in the Carla.ini file worked.

None of them are working now. :/ 
Also, the Carla Settings section has been removed in newer documentation(0.9.4) and it seems CarlaSettings.ini file is simply ignored by the shell script now.

geekhut, 2019-06-04:
> Is it also possible to disable the HUD in the Editor?

Hi have you solved this? I would like to know how.. thanks

Derekabc, 2019-06-17:
Hi, @aayushARM , have you solved this problem and how to run the code without rendering?

tojimahammatov, 2019-11-08:
Hi all. I am using Carla 0.9.6 now, and the following are working without any problem:

pirate-lofy, 2020-06-07:
In linux, just write this command...
`SDL_VIDEODRIVER=offscreen ./CarlaUE4.sh`

siposcsaba89, 2020-07-10:
Hello,

I am trying to disable rendering in carla, version 0.9.9.4 on windows 10. I have already tried with the config script:

./config.py --no-rendering

After that carla window turns to black, but the gpu usage still very high and if I start 

./no_rendering_mode.py

Carla windows start to display the rendered view again. Does anyone have similar issues? Or solutions? Should it work on windows? Thanks in adavance.

mkhansenbot, 2023-05-25:
I see the same problem in version 0.9.12. There isn't a CarlaSettings.ini file anywhere and the `-disable-rendering` flag does nothing. If anyone knows how to disable rendering at startup, please update.

TzabarDolev, 2023-10-04:
Hi, I wrote a blog post about this issue on the CarlaSimBlog, including a code tutorial for off screen rendering.
https://carlasimblog.wordpress.com/2023/10/04/rendering-images-off-screen-in-the-carla-simulator/