carla-simulator/carla #287: What is -carla-no-networking flag?

GokulNC, 2018-03-20:
What does the `-carla-no-networking` option do?
By disabling networking, is it not possible for a client to connect to the server?

Is it the same as setting `UseNetworking=false` in the `CarlaSettings.ini` file?

The comment in the settings file reads:
> If set to false, a mock controller will be used instead of waiting for a real client to connect.
So what does a mock controller here mean?
Where does the controls for the mock controller come from? Are they random?

Thanks :)

marcgpuig, 2018-03-20:
Hi @GokulNC,

> What does the `-carla-no-networking` option do?

It disables networking **regardless** what you have in `UseNetworking` in `CarlaSettings.ini` file.
Useful for a quick test without changing `CarlaSettings.ini`.

> By disabling networking, is it not possible for a client to connect to the server?

Indeed. This way you can drive the car directly from the server without a client but your keyboard.
We use this to test that the server works, for debug or just for showing CARLA to people :)

> So what does a mock controller here mean?

Internally the server is just a controller for the game. The mock controller is a dummy controller that just enables the keyboard input, this way you can use carla without a client.

Regards!

GokulNC, 2018-03-20:
Thanks @marcgpuig 

I tried setting `UseNetworking=false` in my `CarlaSettings.ini`, but eventhough, my client is able to connect to the server and send controls.
Is this a bug?

marcgpuig, 2018-03-20:
We have a known issue #215.
Could be your problem?

GokulNC, 2018-03-27:
No, that's not my problem.
The path I gave is relative and it works and loads `CarlaSettings.ini`

So the problem is not with the file being not loaded.

marcgpuig, 2018-05-03:
Hi @GokulN,

Any progress on the subject?
Unfortunately, I can not reproduce your error :(