carla-simulator/carla #288: Autopilot destination

TairuiWang, 2018-03-20:
Hi,

Currently, I am going to collect my own driving data by 'Autopilot'. I checked the [code](https://github.com/carla-simulator/carla/blob/bb3e66d4421fdedc388b7a456fd9f86cf7bf18e3/PythonClient/client_example.py#L152) and it seems that we don't need to set a destination for it. So, is it possible for me to get driving data along with high level control commands from 'get_next_command'? From your imitation learning project I think it is possible, but I have no idea how to get this.

Another question is how can I know the corresponding indices of way points?

felipecode, 2018-03-20:
Hello @TairuiWang 

You are right, the autopilot from the server side has no 
destination assigned. The data for the imitation learning
paper was collected using client side agent. 

This is currently implemented on the client-side agent branch 
https://github.com/carla-simulator/carla/tree/client_side_agent

You have an example that runs the agent at the PythonClient folder,
it is called "autopilot_control.py".
For it to work properly you have to select the correct town so just run:

python autopilot_control.py -m Town01

Probably it is not working for python3 yet.
You can configure the autopilot behavior by changing the parameters
at carla/autopilot/pilotconfiguration.py

Something like this is needed to have both commands and control

```
# Instance autopilot
self._autopilot = Autopilot(ConfigAutopilot(city_name))
#Instance planner
self._planner = Planner(city_name)

#At your main loop...

# get the control
control = self._autopilot.run_step(measurements, sensor_data,target)

# get the commands

direction = self._planner.get_next_command(current_position,current_orientation,
                                       target_position,target_orientation)
```

Quite experimental, has a few bugs.
Improvements on that would be helpful. 
Probably working on that for 0.8.1 

Cheers

closed 2018-03-22 by TairuiWang

jackspp, 2018-05-08:
Hi @felipecode ,

When i use this  client-side agent branch to collect direction information, the view starts from top left sky,
and keep printing out "Loadding new episode..."

Do you have some suggestions to solve this problem? THanks !

Naeem-996, 2018-09-13:
Hi @felipecode @jackspp,
    I had the same problem @jackspp explained above. Were you able to solve this problem ? Thanks.

KeishiIshihara, 2018-10-25:
Hi @felipecode @jackspp @Naeem-996,
I have the same problem, too. Could you solve this problem?
Thanks.