carla-simulator/carla #289: Changing the weather using a python client

natanielruiz, 2018-03-20:
Is there any way to change the weather using a python client?

Thanks a lot!

marcgpuig, 2018-03-20:
Hi @natanielruiz,

Sure!
You can change the weather and every carla setting that appear on [`CarlaSettings.ini`](https://github.com/carla-simulator/carla/blob/master/Docs/Example.CarlaSettings.ini).

Please take a look at the [`client_example.py`](https://github.com/carla-simulator/carla/blob/master/PythonClient/client_example.py)
https://github.com/carla-simulator/carla/blob/bb3e66d4421fdedc388b7a456fd9f86cf7bf18e3/PythonClient/client_example.py#L43-L53

natanielruiz, 2018-03-20:
Yes I understand. Would it work if I set the settings to something different (thus changing the weather) while the game is playing?

marcgpuig, 2018-03-21:
Well @natanielruiz, this is not currently possible without a "reset".

Anyway, if what you are aiming for is to change the weather each frame, a workaround could be making a blueprint in the editor that does exactly that, from the simulator itself.

Please, give us more details and we'll try our best to solve your problem :)

closed 2018-03-21 by natanielruiz

marcgpuig, 2018-03-21:
@natanielruiz awesome! :open_mouth:

Maybe you could share your solution so people with similar issues can use it!
Regards.

mhusseinsh, 2018-10-10:
@natanielruiz could you please clarify on how did you do that ? I want also to change the weather each frame using the python client, instead of pressing the C button

@marcgpuig any idea ?

chase03, 2020-06-29:
Is possible changing weather parameters in Unreal Engine editor via widget slider? I can't find any interface