carla-simulator/carla #294: why manual control doesn't work

peter-jean1, 2018-03-22:
First all of thank you for your amazing work. i'm working with the compiled version of carla and i only changed the carla directory of PythonClient to follow the changes you made on the project, when i run the python client from manual_controle.py the manual control does not work.

nsubiron, 2018-03-22:
Hi @peter-jean1,

Our GitHub master branch contains already what's going to be version 0.8.0 of CARLA. We have changed few things in the server-client protocol that make it incompatible with previous versions. I'm afraid that if you want to use the latest version of the manual control you'll need to compile CARLA from source, or wait for the coming 0.8.0 release that will be ready in the coming days.

peter-jean1, 2018-03-22:
Hi @nsubiron,
Thank you for your answer, I found a solution except the pygame window which is displayed in black.
Other than that, I see a lot of work on optimizing the graphics quality level.

nsubiron, 2018-03-27:
@peter-jean1 [Release 0.8.0](https://github.com/carla-simulator/carla/releases/tag/0.8.0) is out now.

peter-jean1, 2018-03-28:
@nsubiron Thank you for your tag.

closed 2018-04-03 by nsubiron