carla-simulator/carla #296: Does the new release of carla has mapping coordinates on Town1 and Town2?

LoayMAmin, 2018-03-22:
I just upgraded to the new carla release to use the LiDAR sensors.
However, In the old release, I used the 
x          =             measurements.player_measurements.transform.location.x
y          =             measurements.player_measurements.transform.location.y

Then 

x_pixel = (x + 1643) // 16.43 
y_pixel = (y + 1643) // 16.43 
pixel = Town_image[int(y_pixel), int(x_pixel)]

Which will give me the mapping of the exact pixel in the Town1 from the world coordinates.

Suddenly when I switched to the new Carla release this mapping factor does not work.

Please advice.

nsubiron, 2018-03-22:
The measurements are now sent in meters (previously it was in centimeters) #273. That's probably it.

LoayMAmin, 2018-03-22:
Fixed. Thank you.

closed 2018-03-22 by LoayMAmin