carla-simulator/carla #298: UBT erroe

naveen7v, 2018-03-22:
Hi...trying to build Carla on Ubuntu-16.04

On running Rebuild.sh...the following errors appear

```
/home/nav-u/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/CityMapGenerator.cpp:314:32: error: no member named 'ProjectSavedDir' in 'FPaths'
    RoadMap->SaveAsPNG(FPaths::ProjectSavedDir(), World->GetMapName());
                       ~~~~~~~~^
In file included from /home/nav-u/carla/Unreal/CarlaUE4/Plugins/Carla/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/Carla/Module.Carla.cpp:8:
/home/nav-u/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/DynamicWeather.cpp:24:38: error: no member named 'ProjectConfigDir' in 'FPaths'; did
      you mean 'SourceConfigDir'?
  FilePath = FPaths::Combine(FPaths::ProjectConfigDir(), FileName);
                             ~~~~~~~~^~~~~~~~~~~~~~~~
                                     SourceConfigDir
Runtime/Core/Public/Misc/Paths.h:169:17: note: 'SourceConfigDir' declared here
        static FString SourceConfigDir();
                       ^
In file included from /home/nav-u/carla/Unreal/CarlaUE4/Plugins/Carla/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/Carla/Module.Carla.cpp:29:
/home/nav-u/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Sensor/SceneCaptureToDiskCamera.cpp:15:40: error: no member named 'ProjectSavedDir'
      in 'FPaths'
  SaveToFolder(FPaths::Combine(FPaths::ProjectSavedDir(), "SceneCaptures")),
                               ~~~~~~~~^
In file included from /home/nav-u/carla/Unreal/CarlaUE4/Plugins/Carla/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/Carla/Module.Carla.cpp:36:
/home/nav-u/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Settings/CarlaSettings.cpp:196:48: error: no member named 'ProjectConfigDir' in
      'FPaths'; did you mean 'SourceConfigDir'?
  LoadSettingsFromFile(FPaths::Combine(FPaths::ProjectConfigDir(), TEXT("CarlaSettings.ini")), false);
                                       ~~~~~~~~^~~~~~~~~~~~~~~~
                                               SourceConfigDir
Runtime/Core/Public/Misc/Paths.h:169:17: note: 'SourceConfigDir' declared here
        static FString SourceConfigDir();
                       ^
4 errors generated.
[4/11] Compile CarlaUE4GameModeBase.gen.cpp
[5/11] Compile CarlaMapGenerator.cpp
[6/11] Compile CarlaUE4.cpp
[7/11] Compile CarlaUE4.init.gen.cpp
[8/11] Compile CarlaMapGenerator.gen.cpp
[9/11] Compile CarlaUE4GameModeBase.cpp

ERROR: UBT ERROR: Failed to produce item: /home/nav-u/carla/Unreal/CarlaUE4/Plugins/Carla/Binaries/Linux/libUE4Editor-Carla.so

Total build time: 226.58 seconds (Local executor: 0.00 seconds)

Makefile:600: recipe for target 'CarlaUE4Editor' failed

make: *** [CarlaUE4Editor] Error 5
```
Any help to resolve this please.

Thank you

nsubiron, 2018-03-22:
Hi @naveen7v,

We have upgraded CARLA to Unreal 4.18, this error occurs if you are using a earlier version.

ArghyaPal, 2020-08-16:
Same issue with Unreal 4.18. Any help will be appriciated